theyosh/TerrariumPI #1: Sensor & Relay suggestions

pewi81, 2017-07-10:
Hi
First of all I would like to give huge kudos to you. This is a really neat Terrarium controller that fits my need :-)
I would like to suggest an enhancement though; it would be great if one could add one or more DHT22 Temp & Humidity sensors directly to a GPIO and control in from TerrariumPI, the same goes for a standard relay board which can be controlled from one GPIO on the Pi.
Thanks for a great job

Regards
Petter

theyosh, 2017-07-11:
Hi Petter,

thanks for the compliment. Your suggestions seems cool. It can take a will to add this. I need to get the right hardware and test it.

It also involves some web interface changes for this.

So for the DHT22 I can get some hardware to test with. But the the relay board, I am not sure. Do you have an online example?

Regards
Joshua

pewi81, 2017-07-11:
Hi and thanks for a swift reply!
The relay is a 5V relay board with 2-16 relays which is driven by the 5V pin (and ground) on the Pi, the relays are then each connected to a desireable GPIO to be raised/lowered. The board you link too should be fine! The relay boards I use can be found at Ebay:
http://m.ebay.com/itm/5V-8-Channel-Relay-Module-Board-For-Arduino-AVR-PIC-MCU-DSP-ARM-LW-/172396554144?epid=2068679709&hash=item2823a2a7a0%3Ag%3ADYYAAOSwImRYGpkT&_trkparms=pageci%253A15b5a44b-6628-11e7-9977-74dbd180729f%257Cparentrq%253A314e4a2115d0aa132be552c4ffe93b86%257Ciid%253A1

Best regards
Petter

theyosh, 2017-07-11:
I close this issue and have opened two new ones. So we can discuss per issue.

closed 2017-07-11 by theyosh