theyosh/TerrariumPI #6: Suggestion: Adding 'Cooler' under System environments

pewi81, 2017-07-13:
Im adding another suggestion, I hope Im not to annoying :-) 
I  love TerrariumPI, its great and has so much usefullness and power.

Under 'System environment' there are options to add a heater. It would be useful to be able to add a cooler (turn on fan, cooling device etc.) as well. If a certain max temperature is reached it would be good to have the cooler to automatically turn on until a desirable temperature is reached. This option for the heater to turn on/off at a minimum temperature to raise the temperature to a given value could also be useful.

theyosh, 2017-07-14:
Hi, no problem.
This feature is just 'the heater in reverse'  :)
So I have created an attempt on the branch [cooler](/theyosh/TerrariumPI/tree/cooler). Can you test this? I do not have a scenario for testing this real good. Else I have to merge it without proper testing.

It works that x amount of temperature sensors can be used. When the average of all those sensors will hit the average max, the switch(es) will turn on. As long as the average temperature is NOT below the average minimum, the power will stay on. When the average minimum temperature is reached, the switches will power down, and stay out until the average max temperature is reached.

This is the same way a fridge does its work :D

pewi81, 2017-07-14:
Hi!
I will have a look at it :)
Many thanks

pewi81, 2017-07-14:
Tried to test it but for some reason I can't access the web interface anymore, reinstalled everything but still no access. I have to look into this after work tomorrow.

pewi81, 2017-07-14:
The cooling relay doesn't turn on when the set Alarm temperature is reached. I can't toggle the relay manually either, can't  find the problem. Im using a relay connected to the GPIOs on the Pi.

theyosh, 2017-07-14:
GPIO power switches are not yet implemented. I am waiting on the hardware. I just have prepared the interface for it already.
Sorry for giving the idea that GPIO relay boards should work

PR #7 (2017-07-14): Merge Cooler system
This will add a cooler system to TerrariumPI. Closes #6
closed 2017-07-14 by theyosh