theyosh/TerrariumPI #10: Option for changing to Fahrenheit

Kevinm94, 2017-07-26:
Is there a way to change all the temperature readouts to Fahrenheit? Or have an option in a setting to change it?
Thanks again!

theyosh, 2017-07-26:
Hi, not yet :). As there only 5 countries using Fahrenheit, I did not fully implement it. But I can do that.

But, then I do need to know if your current temperature values are to low / are in Celsius. Because I know that through OWFS, I can also change the value of the sensor to Fahrenheit, and then I should not try to calculate a second time.

For DHT versions I do not know if they can put in Fahrenheit. But because you are asking now, I guess that DHT sensors always give there information in Celsius. Is this correct? Your values are now in Celsius.

The solution would be that I move the option for Celsius or Fahrenheit will move to the general settings page and use it everywhere in the software.

Kevinm94, 2017-07-26:
I believe the current temperature is fairly close, 24c on one and 27c on another both are a keyboard width apart for testing. As for the DHT sensors, in arduino you can make them show Fahrenheit, not sure how it changes on PRi. My current values are displaying in Celsius.
The idea of a general setting and use it everywhere is awesome. 
I also sent you an email regarding relays
Thanks
km

theyosh, 2017-07-26:
First attempt. Looks pretty promising. So you could try to checkout the master branch.

I looked at the Adruino code / Library from Adafruit, and there default is Celsius. So that is what I do and I can just convert them to Fahrenheit.

Kevinm94, 2017-07-27:
Very promising ! I have it loaded and will be able to try it out later today. 
Thanks again !!!

Kevinm94, 2017-07-28:
I was able to try it last night. The only Temp I could get to work was from the Pi. it was showing 37 degrees F. Should be closer to 100F as both my Pi ZeroW and B model run around 98 to 115F during normal operations. 
I am going to try and install a fresh clone to see if it is something I did. I could not get the dht11 settings to save and I was not asking me to enter login info.

theyosh, 2017-07-28:
Hmm, the 37F is correct in a way that 37C is pretty normal for a Pi. So it shows F, but calculates still in Celsius.

Make sure you are pulling the latest master branch. And restart the software. This can be done by login in as pi user and run 'screen -r' This will give a new screen where the software is running. Press Ctrl+C and wait. It will restart in 10 seconds.

Changing from C to F is now done in system -> settings. This should also take care for your sensors. So they do not need an update to get F values.

theyosh, 2017-07-28:
Authentication will only be asked once. As long as your browser stays open, it will not ask again. This is handled by the browser.

Saving sensor data will take some more time now. But still, I can change 3 sensors in 10 seconds. And no errors.

theyosh, 2017-07-31:
I close this ticket, due to no more response, and I can change it to F correctly

closed 2017-07-31 by theyosh