theyosh/TerrariumPI #14: username

50915091Mica, 2017-08-10:
Hi there,
sorry to bother, I am new to the commuity and I got impressed by the project TerrariumPI. I got everything nneeded and started new terrarium. I got through the istallation and the web interface seems working ok. But whenever I try to add a sensor etc. it requires a username and password - no password I have seems to work, can you help me? your project has just all I dream of, i don´t want to stop now..thank you...

theyosh, 2017-08-10:
Hi,

it should be default: https://github.com/theyosh/TerrariumPI/wiki/FAQ#what-is-the-default-login

so try admin / password as default. But change it on the settings page afterwards.

50915091Mica, 2017-08-13:
[image]
Thx a lot, I managed to get to know with the system and seems just the perfect tool I needed. Posting a modet picture of ongoing project... good work! I owe you..

50915091Mica, 2017-08-13:
Just one more thing, is there any chance that your system could support a soil moisture sensor from arduino? I am thinking of how to automatize plant watering by a small pump...thank you...

theyosh, 2017-08-14:
Hi, unfortunately no. I will not make support for Arduino devices. This is not as easy as looks. Because how does the RPI and Arduino talk to each other? That can be done on many different ways. So that would mean that I have to program Adruino also. I have not intention to do so, nor have I time to do so.

So for now, there will be no moisture support.

closed 2017-08-14 by theyosh

50915091Mica, 2017-08-17:
Thanks for explaing, I understand, just one thing, I experience now a new problem - I have 2 DHT22 sensors and one DHT11 sensor and in the "sensor settings" I have troubles adding a new sensor when set to "temperature" modality. When set to "humidity" it works fine, except the DHT 11 - this one doesnt work at all now...
I reviewed the wiring, gpio pins, I´ve even changed resistors (all 4k7), culd you help me where to start ?
Thanks a lot!

tvStatic, 2019-05-19:
Can we get the default username password information in the README.md? I guessed the credentials when I first set it up, and didn't find this or the wiki page until after. It might also be worth put an explicit link to the wiki FAQ in there as well.