theyosh/TerrariumPI #15: Cant add sensors & turn on/off relays

pewi81, 2017-08-13:
Made a new fresh install of latest commit to try out the relay function for GPIO connected relays.
But now I can't add any DHT22 sensors at the settings page for sensors, when submitting the sensor is not saved.
Further, when I add a relay it is not possible to turn on/off the relay, when clicking the relay button nothing happens.

Regards
Pete

theyosh, 2017-08-14:
Pete,

sounds like something is wrong with your pin numbering. Could you try to debug it:

https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

And mail me back the results if there are any errors.

50915091Mica, 2017-08-18:
I have a simillar difficulty, - I have 2 DHT22 sensors and one DHT11 sensor and in the "sensor settings" I have troubles adding a new sensor when set to "temperature" modality. When set to "humidity" it works fine, except the DHT 11 - this one doesnt work at all now...
I reviewed the wiring, gpio pins, I´ve even changed resistors (all 4k7), cables (longest one is 1,5m long) could you help me what to check?
Thanks a lot!

theyosh, 2017-08-18:
I do not have time to look into this until the 20th. So it will take some time before I can test.

50915091Mica, 2017-08-19:
No problem, thank you!

theyosh, 2017-08-20:
Pull the latest commit and both issues should be working.

50915091Mica, 2017-08-22:
thanks! seems working fine now!

theyosh, 2017-09-04:
Closes due to no response of @pewi81

closed 2017-09-04 by theyosh