theyosh/TerrariumPI #16: Non-functioning sensors

50915091Mica, 2017-08-29:
Hi there,
unfortunately I have to ask you for help once more. Without much of a changes in raspberry running TerrariumPI I started to experience problems with sensor reading. Firstly wihout problems, than one DHT11 sensor started to show unrational numbers and I deleted it from the sensor list. Now, after one of reboots none of thw two remaining DHT 22 sensors give any readings. after screen -r command it prints : 

kkk'! Got invalid value 66.900002% in 5.57930 seconds
2017-08-29 13:08:44,282 - ERROR - terrariumSensor - Error updating dht22 humidity sensor 'llllm' with error: float() argument must be a string or a number
2017-08-29 13:09:15,374 - WARNING - terrariumSensor - Error on humidity sensor 'kkk'! Got invalid value 67.199997% in 30.83608 seconds
2017-08-29 13:09:52,857 - WARNING - terrariumSensor - Error on humidity sensor 'llllm'! Got invalid value 3309.100098% in 0.52479 seconds
2017-08-29 13:09:53,648 - WARNING - terrariumSensor - Error on humidity sensor 'kkk'! Got invalid value 69.099998% in 0.52604 seconds
2017-08-29 13:11:23,673 - ERROR - terrariumSensor - Error updating dht22 humidity sensor 'llllm' with error: float() argument must be a string or a number
2017-08-29 13:11:27,027 - WARNING - terrariumSensor - Error on humidity sensor 'kkk'! Got invalid value 3314.800049% in 3.05159 seconds
2017-08-29 13:12:30,035 - WARNING - terrariumSensor - Error on humidity sensor 'llllm'! Got invalid value 77.699997% in 10.62761 seconds

There are now two DHT22 sensors (strangely named I admit), both with 4,7 resistor, connection cables intact. 
The values vary - I don´t think that the value near 70% must be wrong and doesnt seem to me as "invalid"
The values above 3000 - well I read it could be caused by voltage - sensors are connecter to 3,3V and some say that 5V makes it dissappear, what do you think?

Thanks for your reply

theyosh, 2017-08-29:
Some errors I have to fix some code. But the warnings are correct in such a way, that the measured values are outside the given min and max value. This is done to overcome some wrong readings. So the amount of 3314.800049% is invalid due to big. But a value lower of 100 is valid if the sensor has a max of 100.

So, could you check if you have correctly setup your sensors? By default any humidity value between 0 and 100 should be valid.

theyosh, 2017-08-29:
Also this is for me the reason not to use GPIO but I2C or OWFS. That is in my opinion more stable

commit d428552d (2017-09-04): Better warning message when sensor measured values are outside given ranges. Fix #16
closed 2017-09-04 by theyosh