theyosh/TerrariumPI #20: GPIO switches reverse state

penzoiders, 2017-09-07:
I'm using a 8 relay array board over GPIO, VCC and GND wired to the raspberry itself. 
*normal* GPIO numbers used.

Switches are working but in the opposite state: TerrariumPi shows blue status but te led on the relay board is switched on, viceversa when I turn green the switch on TerrariumPi GUI it powers off the relay.

GPIO pointing is ok, status is opposite of expected.

Even on reboot: it starts with all relays on (then, as I've set two relays on the light timer env. rule, it applies opposite rule: off instead of on at given hour)

let me know how can I help to debug these issues. 

And let me thank you for the great project: I looked for something simple and cool like this for years and this is the absolute best project I've encountered so far. deployed istantly.

penzoiders, 2017-09-08:
I discovered that some relay arrays have inverted behaviour.
Mine is one of those.

I've worked around by changing line 92 from terrariumSwitch.py:
`GPIO.output(int(self.get_address()), ( GPIO.HIGH if state is terrariumSwitch.ON else GPIO.LOW ))
`
to
`GPIO.output(int(self.get_address()), ( GPIO.LOW if state is terrariumSwitch.ON else GPIO.HIGH ))
`
that simply worked out.

It would be great to have the option to select "inverted behaviour" when setting up a switch in GPIO mode such as:
[image]

in this way will probably work out for all models of GPIO relay arrays.

theyosh, 2017-09-09:
I was afraid for this. Mine GPIO version is currently working fine. I will add an sort of invert option to the switch.

theyosh, 2017-09-09:
You need to delete the existing GPIO and add a NEW GPIO-Invert in order to get it working. Just changing from GPIO to GPIO-Invert will not work.

closed 2017-09-09 by theyosh

penzoiders, 2017-09-09:
tested. working perfectly.
thanks a lot!

commit 8d8237dd (2017-09-09): Add a new switch type GPIO-invert. Use it when normal GPIO is working in oppisite way. Fix #20