theyosh/TerrariumPI #22: Lock and Logout buttons are not working.

penzoiders, 2017-09-11:
If you press Lock or Logout buttons in the left panel interface nothing will happen and the user remain authenticated (need to clean browser cache + cookies to reset permissions)

tested using 2.7.1

theyosh, 2017-09-13:
That is correct. I do not see a need to use it.
Software is designed for internal use. And therefore a logout is not needed. Also you are not logged in actually. I only made some extra kind or security for making changes.

So I leave it as is. Else you will get a less neat design on the bottom.

penzoiders, 2017-09-14:
If I could suggest one thing.. to wire those 2 buttons to a function that removes the cookie or whatever is used for autentication.
Probably also disabling completely browser caching would help in fixing this (something like https://stackoverflow.com/a/8155121) and in that case logout or lock just calls for a page refresh.

of course nothing really important, but if that's easy will just fix this least-concern security issue

theyosh, 2017-09-16:
Added logout option: https://github.com/theyosh/TerrariumPI/commit/ebe7ce542e7db203d9eb9c5941e1354bb393a926

closed 2017-09-19 by theyosh