theyosh/TerrariumPI #23: Deutsch

vanessa2013, 2017-09-14:
Hi, is it possible to change the whole interface in German?

theyosh, 2017-09-15:
At this moment not. I do not have a German translation. So that should be made first by some one that is a native German speaker. And I am not.

If you want to give a try to translate, here is some information: https://github.com/theyosh/TerrariumPI/wiki/Translations

penzoiders, 2017-09-15:
@vanessa2013 it's not that long to translate, I made the initial italian translation in something like 3 hours while being idle at meetings. go for it!

vanessa2013, 2017-09-16:
I have a small problem I come to my interface no longer

root@raspberrypi:/home/pi/TerrariumPI# python terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 23, in <module>
    terrariumWebserver = terrariumWebserver(terrariumEngine)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 57, in __init__
    gettext.translation('terrariumpi', 'locales/', languages=[self.__terrariumEngine.config.get_active_language()]).install(True)
  File "/usr/lib/python2.7/gettext.py", line 545, in translation
    raise IOError(ENOENT, 'No translation file found for domain', domain)
IOError: [Errno 2] No translation file found for domain: 'terrariumpi'

theyosh, 2017-09-16:
Sounds like the translation files are not yet renamed. Have you placed them at:
`locales/de_DE/LC_MESSAGES/terrarium.po` and `locales/de_DE/LC_MESSAGES/terrarium.mo`?

I saw that the Wiki page was incorrect. That is now corrected.

Could you see if this is the case?

vanessa2013, 2017-09-16:
Hi, So I have taken the .po file from English and the .mo file again.
Saved it should be correct.

[image]

theyosh, 2017-09-16:
I just did the same. No problem here. I did made a change in code last couple days regarding the translations. So maybe your fork is outdated. Could you try your translation with the latest master version?

vanessa2013, 2017-09-16:
I made apt-get update earlier so it should be all up to date, went from one minute to the other not always always offline.

theyosh, 2017-09-16:
I mean the code from github. Not apt.

But, you can also mail your translation files to me: github.com@theyosh.nl and I will add them. Then you can just pull a new version from Github with your translations. Updating is then also a lot easier for later on.

vanessa2013, 2017-09-16:
So I can send it to you yet are not even 20% ready but I can now no longer control whether I have translated everything correctly, I will probably reinstall everything again ...

vanessa2013, 2017-09-16:
But it does not fit any more since I started with the translation :-)

vanessa2013, 2017-09-16:
How can I reload github again without reinstalling the whole rasp?

theyosh, 2017-09-16:
Ok, not sure who well you are familiar with Git. But here is a way that is pretty straightforward. 

Make a backup of your current translated files: `locales/de_AT/` to your desktop.
Delete the directory TerrariumPI on the PI.
Clone the repository again from github on your PI.
Copy back the translation files to location `locales/de_AT/`

This will give you a clean version with the latest code with your translations.

Normally, you have created a fork of my code. But I see you have not done that. So you can make a translation, but you cannot add them to my code. But that is not a problem right now. As long you have the translation files, the can be added.

vanessa2013, 2017-09-16:
What do you mean by fork
Sorry, I can only translate your text with Google ...

theyosh, 2017-09-16:
Fork: https://help.github.com/articles/fork-a-repo/

But forget that. I will make it easy for you. If you just sent your translation by email. I will add them. Then you can just download a new version here from Github and put that on your Pi. After that, you can just continue with translating.

vanessa2013, 2017-09-16:
[image]
Not nice :-(

vanessa2013, 2017-09-16:
I have sent you an email

theyosh, 2017-09-16:
Yes. I have added them to the code.

If you download the latest version with this link: https://github.com/theyosh/TerrariumPI/archive/master.zip

On your Raspberry PI remove the old code, and upload the new code. And start it.

Also, clear your browser cache. My software use some heavy caching for speed. But after an update, you have to clear your cache.

theyosh, 2017-09-16:
Sorry, above is incorrect. This will not have the webinterface :(

So to get a clean version do the following:

login in to Raspberrypi with SSH with user pi
run command: git clone --recursive https://github.com/theyosh/TerrariumPI.git
enter directory TerrariumPI and run ./start.sh

This will dowload the latest version with the webinterface.

vanessa2013, 2017-09-16:
[image]
Und wieder ein Problem :-(

theyosh, 2017-09-16:
You are running it as root. That is not good.

1. Login with SSH as user pi.
2. cd Terrariumpi-master
3. ./start.sh

done.

vanessa2013, 2017-09-16:
Ok, what is currently the master or master without

theyosh, 2017-09-16:
Ok, it is getting late. If it is all not working. Here are once the steps to install it again.

1. Login with ssh as user pi
2. Remove all old code: `rm Terrarium*  -Rf`
3. Download new code: `git clone --recursive https://github.com/theyosh/TerrariumPI.git`
4. Enter the code directory: `cd TerrariumPI`
5. Start the server: `sudo ./start.sh`

This should do it. All commands are `in color`

theyosh, 2017-09-16:
The contents of the zip is different then a git clone version. That makes it confusing

theyosh, 2017-09-16:
And tomorrow there is another day :)

vanessa2013, 2017-09-16:
That's true, wish you a good night

theyosh, 2017-09-17:
Hi,

have you already tried: https://github.com/theyosh/TerrariumPI/issues/23#issuecomment-330001735

That should give you a clean version with German translation you sent me

vanessa2013, 2017-09-17:
Hi, momentarily everything works at the interface, however, he does not take me any translation into the interface

theyosh, 2017-09-17:
Ok, a few questions:

1. Can you choose your German language from the settings page?

2. If you set to German and save the settings does it give an error or ok message?

When changed your language, you have to reload the hole webpage in your browser. If it is not in a different language, use Shift + Reload page (bypassing caching). This should reload the interface in the correct language.

vanessa2013, 2017-09-17:
[image]
So as here New Admin password is written in German but in the interface still in English, I have raspberry and interface reloaded ...

theyosh, 2017-09-17:
Ok, that is somehow correct. But first, I see German translations. Hilfe is Help. I can read a bit of German. :)

But when you make new translations you have to follow the following steps.

1. Save in PO Edit. This creates new translation files
2. Restart Terrarium PI (can take about 30 sec) https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi
3. Clear browser cache and reload webpage

Translations will only be loaded during startup of TerrariumPI. So updating translation while running will NOT update the interface. And due to caching, 'old' translations are shown. Clear browser cache and it should be ok.

vanessa2013, 2017-09-17:
Ok, thank you with cache empty it has worked, I sign up when I am finished with the translation.

vanessa2013, 2017-09-17:
[image]

The only thing since the translation is no more is the animation in the weather.

Is it possible to use Wetter.at, because OpenWeaterMap with me is quite inaccurate

theyosh, 2017-09-17:
For weather data you can also use:

https://www.yr.no/place/Austria/Vienna/Simmering/

The missing weather animations are fixed in a newer version. So please ignore for now.

vanessa2013, 2017-09-17:
Hello, since now the interface shows me always offline, but the Rasp has already started again ...

vanessa2013, 2017-09-18:
Hello, do not you know what that could be?

theyosh, 2017-09-18:
No, because I do not understand what the problem is. 

Could you try the following: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi and see if there are errors or other messages?

Or is the website ok, but only the indicator on the webinterface not working. Please be more specific.

vanessa2013, 2017-09-18:
The help has not helped me, I have sudo git clone --recursive https://github.com/theyosh/TerrariumPI.git loaded and then sudo ./start.sh but I come not at all on the Rasp, and yesterday stood I'm always on me

theyosh, 2017-09-18:
I am sorry, but how do I know what is wrong if I can't see what you are doing. You had a working version, because you asked for Wetter.at support (which I can't). So it was working.

I do not know what you have done afterwards. Nor do I know what error you have right now, because you are not clear. So please be more precise.

Why did you do another git clone? I do not understand what you are doing. Also you English is not always clear what you mean.

vanessa2013, 2017-09-18:
Sorry for my english but I can unfortunately only translate everything about google :-)
Can you see what I mean by TeamViewer?
I have made a new clone because I was always offline in the interface no matter what I tried.

theyosh, 2017-09-18:
I got teamviewer. So we could try that.

vanessa2013, 2017-09-18:
ok i have sent you a mail

vanessa2013, 2017-09-19:
Hello, have you once again that I briefly help with the translation, then I'm done

theyosh, 2017-09-19:
Sorry. I was really busy. With teamviewer again? Sent the invitation and I will have a look

vanessa2013, 2017-09-20:
Sorry, I just fell asleep yesterday, can we do this today ?, if you have time just write me.

vanessa2013, 2017-09-20:
[image]

theyosh, 2017-09-20:
Hmm, looks like some wrong config settings. I can see if I can fix it if you send Teamviewer invite

theyosh, 2017-09-20:
Sorry, but tonight I am not at home. To fix the above error, you could try the following:

Login to your pi with user pi
go to the directory: cd TerrariumPI
remove settingfile: rm settings.cfg

And restart the software https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-restart-terrariumpi or restart the hole PI.

vanessa2013, 2017-09-20:
Hi, I've been able to fix it at the moment, but if you have time tomorrow please look over by TeamViewer.

theyosh, 2017-09-21:
Hi, first I have a meeting (again). But at around 22:00 I should be available for about 30 minutes

vanessa2013, 2017-09-21:
Hi, ok then it goes with me today unfortunately not but egla tomorrow is also a day :-)

vanessa2013, 2017-09-24:
Hi, If you would have time in the evening would be cool, then I can today the German translation and make you send.

vanessa2013, 2017-09-24:
?

theyosh, 2017-09-25:
Euh, I was not at home this weekend. So I am not always available. Please keep an eye on you mail tonight. I hope to have some time at 22:00 hours. I will send an notice when I am available

theyosh, 2017-09-25:
Yesh... I have time now

vanessa2013, 2017-09-26:
Hi, sorry I slept yesterday, if you have time today please contact

theyosh, 2017-09-26:
I am available now. Teamviewer again?

vanessa2013, 2017-09-26:
I sent you the data by mail

vanessa2013, 2017-09-26:
Did you get my mail?

theyosh, 2017-09-26:
No? That is strange... could you try again? github.com@theyosh.nl

vanessa2013, 2017-09-26:
Ok, I've sent it to you again please tell me

[2 more replies]

commit fa6eb733 (2017-09-26): First final version Germand translation. Close #23
closed 2017-09-26 by theyosh