theyosh/TerrariumPI #24: Cant change language

pewi81, 2017-09-16:
I installed the latest update and now I can't change language to English.
There is no options pin the Settings page to change language.

theyosh, 2017-09-16:
Euh, could it be browser cache? Clear it and try again. Also make sure you have restarted the software after the update.

Else I need more information, because I can not reproduce.

penzoiders, 2017-09-16:
It's 99% cache issue. I had similar issue when switching to Italian. Even after restarting the software (screen -r and CTRL+C) the new language wasn't showing. I had to completely clear browser cache. Probably best option is to set "disable cache" at the application level to avoid these UX issues.

theyosh, 2017-09-17:
Closing this. As we are able to add and change languages. Probable cause is browser caching

closed 2017-09-17 by theyosh