theyosh/TerrariumPI #25: Trouble in configuring a DHT22

penzoiders, 2017-09-24:
I've configured a DHT22 sensor using pins 17 (3.3v) 18 (gpio) 20 (ground).
configured through GUI and did not come up with a valid monitoring.

[image]

this is what is thrown at `screen -r`

``` 
2017-09-24 09:50:02,587 - ERROR - terrariumSensor - Error updating dht22 humidity sensor '' with error: float() argument must be a string or a number
2017-09-24 09:50:47,848 - ERROR - terrariumSensor - Error updating dht22 humidity sensor 'DHT22' with error: float() argument must be a string or a number
```

the sensor is one of these with little PCB with embedded resistor convenient pins for GPIO hook up.
https://www.amazon.it/gp/product/B01DB8JH4M/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

To get the server working again I had to remove the sensor configuration. Upon a CTRL+C restart in the screen session I got this output:
```
  File "/usr/local/lib/python2.7/dist-packages/gevent/greenlet.py", line 536, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 211, in __engine_loop
    self.sensors[sensorid].update()
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 207, in update
    humidity, temperature = self.sensor.read_retry(terrariumSensor.valid_dht_sensors[self.get_hardware_type()], self.sensor_address)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 81, in read
    return platform.read(sensor, pin)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi_2.py", line 29, in read
    result, humidity, temp = driver.read(sensor, int(pin))
```

any clues?

theyosh, 2017-09-25:
Sounds like wrong saved config. Can you put your current settings.cfg here?

also, when saving the sensors, can you open you webconsole and see what is being post. Use network tab to see the PUT commands.

theyosh, 2017-09-26:
Ok, it looks like there is a different problem. 

@penzoiders are you running this as root user pi?

@penzoiders and could you see if the user pi is member of the GPIO group? cat /etc/group -> output??

I do not have this issue right now. So in order to replicate I have to install a clean PI. But that can take some time

penzoiders, 2017-09-26:
Hi @theyosh 
I'm using latest 2.7.1 from github and I'm not running as root. It is a completely standard as pi user (that is member of gpio group, my switches are working so far).

```
[sensor1a23ed0036b31d07d66b86cb51ac0e47]
address = 18
alarm = True
alarm_max = 80.0
alarm_min = 30.0
hardwaretype = dht22
id = 1a23ed0036b31d07d66b86cb51ac0e47
limit_max = 95.0
limit_min = 20.0
name = DHT22-1
type = humidity
```
this is the output of the network tab in chrome, after 1,4 minutes it just gives a 200.. the main issue is that the value is `-1` and this blocks the server until I remove the config for the DHT22.

maybe my sensor is bad?

[image]

penzoiders, 2017-09-26:
I saw you updated the install script... should I upgrade the Adafruit_Python_DHT stuff? If so can you tell me how this is done without screwing up things? thanks

theyosh, 2017-09-26:
Just made a small code change. Not sure if it will fix, but could you pull the latest code from the master branch and restart it? See if it makes a difference?

If not, try running the install script once again. It should update to the latest version of all needed modules if there are updates. This to make sure we are all working with the same software versions.

penzoiders, 2017-09-26:
Updating now `git pull` + `sudo ./install.sh`.
I'll update and reboot the pi's raspbian. just to make sure that I don't have something stuck from elsewhere around (with systemd it's never known).

penzoiders, 2017-09-26:
Niet.
Meanwhile updated to 2.8 and restarted again.
no luck.
This is the ouput in the screen session upon CTRL-C restart when not setting limits:
```
2017-09-26 22:39:07,308 - ERROR - terrariumSensor - Error updating dht22 humidity sensor '' with error: float() argument must be a string or a number
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 40, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 2690, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 159, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 747, in set_config
    update_ok = self.set_sensors_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 353, in set_sensors_config
    sensor.set_alarm_min(sensordata['alarm_min'])
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 298, in set_alarm_min
    self.alarm_min = float(limit)
ValueError: could not convert string to float: 
```
When setting limits just:
```
2017-09-26 22:41:32,657 - ERROR - terrariumSensor - Error updating dht22 humidity sensor 'dht22' with error: float() argument must be a string or a number
```

I think I'll buy another sensor and test again, maybe this is faulty (data seems ok when saved in settings, probably output is weird so it throws errors at the DHT22 "driver").
I share here the GPIO hookup for that sensor (as described in first post):
[image]

theyosh, 2017-09-26:
Hmmm, I have to think about this. So this will take some time. It could be some with the hardware. There are problems converting some values to float.
You config is ok. So then it should be a bad read out of the sensor. So I have to make some more checks to see if there are errors. And I think you should try different hardware.

Or, there are also some simple python scripts on the internet to just to test a single dht22 sensor. Google for python and dht22

So you can at least test you hardware more easily.

penzoiders, 2017-09-26:
Mmmhh.. the sensor seems ok. I used the example provided in Adafruit folder. not even had to download something else:
```
pi@terrario:~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 22 24
Temp=23.2*  Humidity=57.1%
```
even without root
```
pi@terrario:~/Adafruit_Python_DHT/examples $ ./AdafruitDHT.py 22 24
Temp=23.2*  Humidity=57.1%
```

is it maybe that `*` and `%` that screw up things in the scripts (just wondering, sorry if I don't take a look but today I'm kind of burn out)

cheers

penzoiders, 2017-09-26:
I found out. I had to use the raspberry GPIO numbering not the BMC one.
now looks ok (it was on pin 18, so I set address to 24) look here: 
https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png

Sum up: for the Switches (as GPIO relay board) I had to use the pin number, for the sensor the raspberry GPIO naming scheme.

see: actual value is now 57
[image]

penzoiders, 2017-09-26:
Just a note on this (happens also when "playing" with switches settings on the "show switches" tab): after changing settings I cannot see in the "show sensor" tabs anything.
This fixes out only by completely wiping out browser cache (or in some days by itself).. maybe forcing absolutely no cache for the application on server side may help.

Data is there and collected properly in main dashboard, but not in the senors tab:
[image]

[image]

vanessa2013, 2017-09-26:
So with me it still does not work
[image]
[image]

penzoiders, 2017-09-26:
@vanessa2013 
this will help out: https://it.pinout.xyz/
try to change the Address from 29 (pin number) to 5 (BCM pinout address for that GPIO pin).
and the magic will happen (hopefully)

cheers.

vanessa2013, 2017-09-26:
Hi, no unfortunately I do not already use pin 29/5. I still have some software to load the pi?

theyosh, 2017-09-26:
If that does not help run the following commands on your pi:

`sudo addgroup gpio`
`sudo usermod -a -G gpio pi`

and restart your pi (is the easiest way to do it)

penzoiders, 2017-09-26:
> Hi, no unfortunately I do not already use pin 29/5. I still have some software to load the pi?
in the screenshot you set pin 29 to Address. so I supposed you made my same mistake of wrong pin numbering. send a pic with of the board and mark the GPIO wire you use for data and we'll check this out.

vanessa2013, 2017-09-26:
@theyosh The tip did not help
@penzoiders I tried it with 5 and with 29 both did not bring anything
[image]

penzoiders, 2017-09-26:
@vanessa2013 looks like (I guess since I cannot follow the wires out of the pic =) you have powered the DHT22 with the 5V (physical pin 2), you should power it using 3.3V (either physical pin 1 or 17).

re wire and reboot maybe.

vanessa2013, 2017-09-26:
[image]
No unfortunately it did not help, no idea where the problem can lie. :-(

vanessa2013, 2017-09-26:
[image]

penzoiders, 2017-09-26:
Test the sensor
```
cd /home/pi/Adafruit_Python_DHT/examples
./AdafruitDHT.py 22 [your_BCM_pin]
```
try with `./AdafruitDHT.py 22`
and see the output

if you have permission issues try with `sudo` in front of that and add (as @theyosh suggested) the pi user to gpio group (do a reboot for sake of lazyness to take effect, or stop completely the screen session 2x CTRL-C, logout, login, `cd TerrariumPi` and `sudo ./start.sh`)

vanessa2013, 2017-09-26:
[image]

penzoiders, 2017-09-26:
@vanessa2013 d'ho
without square brackets, and use BCM.
`./AdafruitDHT.py 22 25`

vanessa2013, 2017-09-26:
[image]

I tried it now with ./AdafruitDHT.py 22 25 but unfortunately comes nothing

penzoiders, 2017-09-26:
mmm.. try to update everything (TerrariumPi and libraries) and reboot, then test again:
```
cd /home/pi/TerrariumPi
git pull
sudo ./install.sh
```
double check the pin you're using matching this scheme: and use the BCM number: https://pinout.xyz/

vanessa2013, 2017-09-26:
[image]
Hmm he does not take the time zone

penzoiders, 2017-09-26:
Timezone should not be an issue, not for this issue for sure.

And if the pic you've sent is still valid the command to test is:
`./AdafruitDHT.py 22 5`

vanessa2013, 2017-09-26:
No the picture is no longer up to date now it has been given to bin 25, unfortunately has not helped so far, can it be that my relay is gpio?

penzoiders, 2017-09-26:
try to wire as I did, just to replicate a working setup.
update with the instructions above, wire like this:
[image]
reboot
try with 
```
cd /home/pi/Adafruit_Python_DHT/examples
./AdafruitDHT.py 22 24
```

vanessa2013, 2017-09-26:
[image]
No is unfortunately not

penzoiders, 2017-09-26:
ok, post the output of
`cat /etc/group | grep gpio`

vanessa2013, 2017-09-26:
[image]

penzoiders, 2017-09-26:
or try, just to stick to the error (just a test)
`./AdafruitDHT.py 22 GPIO1_24`
if this works something is weird with your defaults

vanessa2013, 2017-09-26:
[image]

Nop

penzoiders, 2017-09-26:
look at the error 
try 
`sudo ./AdafruitDHT.py 22 GPIO1_24`

if this works try to set in Address in TerrariumPi `GPIO1_24`

vanessa2013, 2017-09-26:
[image]

Failed to get reading. Try again

penzoiders, 2017-09-26:
may I suggest a clean reflash of raspbian and start from scratch.
It seems that your system is missing something.
get this, https://downloads.raspberrypi.org/raspbian_lite_latest
unzip, flash on the sd card, configure proper network, timezone and ssh access.
reboot
```
sudo apt-get install git
git clone --recursive https://github.com/theyosh/TerrariumPI.git
cd TerrariumPI
sudo ./install.sh
```
reboot

retry with 
```
cd /home/pi/Adafruit_Python_DHT/examples
./AdafruitDHT.py 22 24
```

if that works go and set up the rest in the TerrariumPi gui.
need to sleep now. cheers.

vanessa2013, 2017-09-27:
[image]
Yes, thank you now it works

penzoiders, 2017-09-27:
Horray!

vanessa2013, 2017-09-27:
For this I can access my interface no longer only via SSH

vanessa2013, 2017-09-27:
[image]
Something does not fit the humid

penzoiders, 2017-09-27:
if you don't attach an HDMI screen at boot you may not have access to the console.
raspbian lite doesn't have a GUI, only command line.

did you fill all the upper/lower limits and alarms for the sensor using integer numbers?
can you post your `settings.cfg` file?

PS: please copy paste the text here, don't send screenshots: if you want to paste console text in github the best thing is to use markup to display it correctly use 3 grave accents to "open" and "close" the paste  ```  (https://en.wikipedia.org/wiki/Grave_accent)

vanessa2013, 2017-09-27:
Hi, I use the Raspian Deskop version, yes I have everything filled with integers except for the electricity price since I entered 0.10, how do I find these settings.cfg, and what should I then make the translation is unfortunately very bad

penzoiders, 2017-09-27:
`cat /home/pi/TerrariumPi/settings.cfg`

vanessa2013, 2017-09-27:
[image]
[image]

Hmm now it is suddenly again, no idea why, but me shows the DHT22 always only 10 degrees and air 0%, something runs wrong with me

vanessa2013, 2017-09-27:
[image]
 I think there is something wrong

vanessa2013, 2017-09-27:
[image]
Again the same problem as yesterday, man that annoys

vanessa2013, 2017-09-27:
Humidity=3307.3% :-)

theyosh, 2017-09-27:
I saw somewhere you had entered a power price of 0,10. This will not work. And should use 0.10

The comma is not working for decimals. Use a dot.

It is a bit strange because all numbers are using a comma for decimal separation due to language choice. But entering data it should always US format. So use a dot for decimal separation

penzoiders, 2017-09-27:
I think that an hint on the field label can avoid issues for now.
Definitive solution may be to parse the inputed field and process it to keep into account only the last separator, or the first from the right, and replace it with a dot if it is a comma. Thrashing all the remaining separators in the leftmost part of the number (think someone using a dot to separate the thousands).

theyosh, 2017-09-27:
I think most people do know this, or can figure it out. 
And there are input parsers based on language. Will have a look into that

penzoiders, 2017-09-27:
> I think most people do know this, or can figure it out.
And there are input parsers based on language. Will have a look into that

For sure it's not rocket science, but I know that if an end user has the opportunity to set something wrong, this will happen. =)

I think that an hint in the label field is more than enough for now.

A parser that takes locale into account it's a good idea, but maybe more important is to avoid platform crashing (better set a null value, or valid default if the configuration field is proven to be wrong in format).

vanessa2013, 2017-09-27:
Thanks @theyosh , unfortunately, the temperature is not synonymous I have here 24 degrees and 14 degrees is displayed in the interface. And with 3307.3% humidity I can leave the comma away as I want I still come on no real value and when I move the comma I come to 33% and that is not true.

vanessa2013, 2017-09-27:
2017-09-27 21:01:03,864 - ERROR - terrariumSensor - Error updating dht22 humidity sensor 'Luft Test 2' with error: float() argument must be a string or a number
2017-09-27 21:01:41,770 - ERROR - terrariumSensor - Error updating dht22 temperature sensor 'Temp Test 1' with error: float() argument must be a string or a number

I do not know what it means but maybe it helps with the error search

theyosh, 2017-09-28:
Ok, I close this issue now, because the initial problem is solved. And I get the feeling that there is no software problem now. And therefore there is nothing more to do in code.

closed 2017-09-28 by theyosh

vanessa2013, 2017-09-28:
Ok, can you please explain to me why the sensor always indicates false values (eg 14 degrees I have 27 degrees in truth), humidity does not show me at all is always at 0 or 3.307%

What still strikes me is that in a change eg. Sensor add or remove always extremely long the interface loads.
The following I have already tested: Completely new Raspi OS, new screen, other DHT22 sensor.

Thank you for your help

vanessa2013, 2017-09-28:
Ok, can you please explain to me why the sensor always indicates false values (eg 14 degrees I have 27 degrees in truth), humidity does not show me at all is always at 0 or 3.307%

What still strikes me is that in a change eg. Sensor add or remove always extremely long the interface loads.
The following I have already tested: Completely new Raspi OS, new screen, other DHT22 sensor.

Or should I open a new ticket for this topic?

Thank you for your help