theyosh/TerrariumPI #27: A way to avoid chrome caching issues

penzoiders, 2017-09-27:
I know I notified this in another issue https://github.com/theyosh/TerrariumPI/issues/25, but maybe to keep things clear it's better to track it on a new issue.

I found that when using chrome (or any other heavy caching browser) if you configure a new switch or a new sensor it will not appear in the "status" tab for sensors or switches until browser cache does not expire.

I found a workaround using the method of "hidden refresh menu" stated in this superuser.com post https://superuser.com/a/722548 "Empty Cache and Hard Reload"

this when done completely solve the issue. Since the whole application is not that heavy it may be a good idea to automatically drop caches at page load (don't know if it's possible), or to set a cache expiry to a very low level.

@theyosh do you think it's something that can be set in gentelella?

theyosh, 2017-09-28:
Here we have a different opinion about this. I will explain why I think it is a bad idea to change the caching.

When I started creating this software on a Pi 2 with a SD 4 type card, it took very long load all the HTML files and javascript. It was really slow. So by adding caching for static files, it became much faster. And so I have created different caching setting for different type of files. This took some time to find out the best performing settings.

Secondly, the amount of changes you make and having this issue versus the amount of days you will not make changes is somewhat 1 to 10000 ? In other words, you will change in the begin a couple of times and then never again. I have made my terrarium, configured it, and now I did not need to change it for 3 months. Because you will not add or remove sensors when the setup if finished.

Therefore I think there is nothing to gain by disabling the caching. Because when adding new sensors during setup, you know that you have to clear the cache. If this has to be done 5 times to get everything configured it is not much work to clear your browser.

Alternatively you can run you browser in private modus, which will never cache data. When configured, you can use the normal modus again.

So for me there it is not worth to spent time to make code for something that is only happening for 5% of the time. And there are ways to disable caching in the browser if wanted.

penzoiders, 2017-09-28:
I see your point. And I have no issues in handling the cache cleaning. It was just about new or not technical users.

I don't think we need to develop a software side solution to this, maybe just a Troubleshooting section of the online help will do the job.

As soon as I find out how I'll prepare the extra page on my fork, or, if you can provide a new "tab" under the help section called Troubleshooting We can add those little extra tips that are the common faults of a new user. Eg: "I set the language but it does not apply" --> screen -r + CTRL C, or "I added a switch but my switch status page is screwed up" --> hard clean cache procedure for major browsers.

I don't say it has to be all automated, many issues on a small scale project like this can be solved with a bit of user-friendly docs (which are moslty already in place, and this is great btw).

thanks for taking time to share your view.

cheers

vanessa2013, 2017-09-28:
Hi, ich hätte eine andere Frage hat von euch jemand bei der Sensoradresse die Beschreibung (Test zur Hilfe) geändert bei mir ist sie nicht mehr Deutsch sondern Englisch, und lässt sich auch nicht mehr übersetzten...

vanessa2013, 2017-09-28:
Hi, I would have another question has changed from you someone at the sensor address the description (test for help) with me it is not more German but English, and can not synonymous translated ...

penzoiders, 2017-09-29:
> Hi, I would have another question has changed from you someone at the sensor address the description (test for help) with me it is not more German but English, and can not synonymous translated ...

I really don't understand the question. sorry.

vanessa2013, 2017-09-29:
[image]
My question is whether you have changed this description?
Because I can not translate them into German

penzoiders, 2017-09-29:
You need to update your translation file to translate the new/modified labels: https://github.com/theyosh/TerrariumPI/wiki/Translations

theyosh, 2017-09-30:
It can happen that some parts of the interface will get new text. And then the translation has to be done again. This is the only way to do it. At this example, it is only one sentence.

Sorry if that will cause some extra work for you.

theyosh, 2017-10-04:
No more progress. Also caching will have some pros and cons....

closed 2017-10-04 by theyosh

vanessa2013, 2017-10-04:
Hi, sorry my last days were a bit stressful thank you for your tips I will make it new, nice evening yet.