theyosh/TerrariumPI #28: Heating control via sensor temperature

vanessa2013, 2017-09-27:
How is it possible to regulate my heating when, for example, 24 degrees. Do I have to adjust this somewhere? The heater must know that they turn on at 23.5 degrees and turn off at 24.5 degrees, but this is only a sensor but unfortunately I have not found anything yet.

theyosh, 2017-09-28:
Yes, this is possible with one sensor.

1. Make sure you have setup a switch that can control power on your heating.
2. Configure the used temperature sensors with alarm min value of 23.5 and alarm max 24.5. If that does not work, use 23 and 24 or 24 and 25.
3. Go to the Environment page, and choose by heater the Sensor mode and select the configured switch and sensor.

Save it.

That should do it. When the sensor goes below the alarm min value, it will put on the heater. When the alarm max is hit, it will turn off the heater.

theyosh, 2017-09-28:
This should also be explained at the Usage pages.

Go to About -> Usage in the menu left
Then tab 'Environment' and on the right side chose 'heating'. If the Google translation did it's job, should explain there also how to setup your heater

vanessa2013, 2017-09-28:
Hi, ok thanks.

closed 2017-09-28 by vanessa2013