theyosh/TerrariumPI #29: After the Initial Install, I can't access the web interface

jbrooks4, 2017-10-04:


penzoiders, 2017-10-04:
A bit more of detail would help

penzoiders, 2017-10-04:
Did you access it on the correct port? 

http://[your-raspi-IP]:8090

Not default TCP 80 (for that you can always proxy with NGINX or other web server)

Check your firewall on raspberry (if in place), if you ever did one: a reboot is needed.

jbrooks4, 2017-10-04:
Yes I used to correct port , and no firewall on. I will wipe it and do another clean install and see what happens! Is there anything extra to do after the TerrariumPI install ?

penzoiders, 2017-10-04:
No, a clean raspbian installe, git pull recursive the repo, install, reboot, enjoy.

jbrooks4, 2017-10-04:
Okay I will post back when I'm done. Cheers !

theyosh, 2017-10-04:
Update the installation and update steps. Now you can just copy paste the needed commands

https://github.com/theyosh/TerrariumPI#installation

jbrooks4, 2017-10-04:
Hmm I was trying the Full version not the Lite could that have been my problem ? Reinstalling lite right now !

penzoiders, 2017-10-04:
Full should not be a problem, lite is preferable for the use.

jbrooks4, 2017-10-04:
I did exactly those commands on the Full version with no luck. :(

theyosh, 2017-10-04:
Can you try this: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Not sure how to copy paste the text you will see. But give it a try...

And the full version should also work. But takes more time to download and burn to image

jbrooks4, 2017-10-04:
Worked this time on the clean install of Lite ! Maybe I canceled something by accident being impatient the first time ?

jbrooks4, 2017-10-04:
One last question for now :P What is the default username and pw for the web interface ?

jbrooks4, 2017-10-04:
Is there a schematic anywhere for all the wiring ?  For http://fritzing.org/home/   ?

theyosh, 2017-10-04:
Default user and pw: https://github.com/theyosh/TerrariumPI/wiki/FAQ#what-is-the-default-login

There is no wiring schema's due to the fact my software does support all different kinds. If you have your TerrariumPI running, go to menu heading 'about' -> 'Help' -> 'Hardware'. 

You can then google for more information about the used sensors and switches.

I am not able to support hardware issues. I do expect some kind of 'handyness' with hardware and google :D

jbrooks4, 2017-10-04:
No problem , I am pretty handy with the sensors, my other question is about a few other sensors I have and if it will support them. I have https://goo.gl/MocDDB ,  https://goo.gl/ghaob3, https://goo.gl/Xum2vP

theyosh, 2017-10-04:
Nope... sorry. The first one I am working on. But the problem with those sensors is that they require an analog input. And a Pi does not have those. For a Pi you need sensors with digital data.

In order to support analog input sensors, there is a need for a extra board that converts analog to digital. And that is a bit harder to support. And also, it requires more hardware. So for now, I cannot support sensors with analog input

theyosh, 2017-10-04:
But, I guess the problem with the webinterface is sorted out? So I can close this issue?

For your sensors I am afraid that it will not be soon. For me there is not that need to add more sensors. And also, if I do not own the specified sensors, it is a bit hard to make code for it...

jbrooks4, 2017-10-04:
Okay no problem ! The Ph one would be the biggest one I wanted! I was about to ask about the conversion boards :P

closed 2017-10-04 by theyosh