theyosh/TerrariumPI #30: Problems with relay card and DHT 22

vanessa2013, 2017-10-05:
Hi, I have a problem with my Relaiskarte (8x relay), every time I switch on one or all relays over the interface, they are automatically switched off after approx. 30 sec from the interface without which I do something or enter, my second problem is that I only one of three DHT22 sensors to run, one shows me correct values the others are not recognized, can it be that there is a problem with the 5V GPIO? Because I have all three DHT22 sensors on a 5V GPIO with a splitt cable. Thanks for your help!

vanessa2013, 2017-10-06:
Hi, no one has any idea of you?

vanessa2013, 2017-10-06:
What also comes when the relay switches off itself the button in the interface green

theyosh, 2017-10-06:
I would like to ask to make 1 issue for 1 problem the next time. That makes it easier to communicate.

For now:
1. I cannot help much with your hardware. As there is a language barrier (dutch vs german through english) it is very difficult to explain. And I do not have all the knowledge about all hardware. So I think you should start with 1 sensor and test it according to: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-test-a-dht11-dht22-or-am2303-sensor . If for what reason you hardware is not working you have to figure it out with Google and German blogs. That is what I can do about your hardware issues.

2. That you switches go off after 30 seconds could be because you have selected them in the Environment page. If you select switches with either cooling, heating or spraying. And if you have selected timers, you cannot switch on manually. All switches that are used in the Environment page are controlled by the Pi. So look at your setup which switches are configured

theyosh, 2017-10-06:
Again, you comments are unclear.

vanessa2013, 2017-10-06:
Hi, I wrote above that my relays switch off itself without the I do something, the relays are not integrated in any heating or cooling. And then I wrote still if it is a problem if I have 3 DHT22 sensors at the same 5 Volt gpio

theyosh, 2017-10-06:
Ok, I can't do anything with this. I do not know what kind of relay board you have. So again, there is a lot of different hardware that could work with my software. But there is no way to test them all. So either provide with more information about brand and type. Mostly the hardware supplier does have some example code to test the device. Try that first.

For you DHT22 sensors, it is clear there not working. The only thing I can say about it: Google for it. I do not have that knowledge.

I want to emphasize that this software is developed with in mind that when using it, you are familiar with a bit programming at least able to read code. Familiar with the sensors and switchboard you want to use. It is impossible for me to support everybody and anything.

There is also a log file that could explain some actions that are happening. Take look at the file: log/terrariumpi.log . It should write loglines when the switches are changed in state.

vanessa2013, 2017-10-06:
Ok, thank you has done it were contact errors with the 5 Volt GPIO

closed 2017-10-07 by vanessa2013