theyosh/TerrariumPI #34: Translate

vanessa2013, 2017-10-22:
What can I do if I can not find all words or phrases in Poedit and I can not translate them? Thx

vanessa2013, 2017-10-22:
Something else, I can not change the password, it will not be saved

theyosh, 2017-10-24:
I can change my password. So I cannot reproduce your problem. Please use normal characters and numbers first.

If you are missing translations, please make a screenshot of what should be translated. Else I do not understand your question.

I can say that all Weather forecast is non translatable due to the fact it comes from an external API.

vanessa2013, 2017-10-25:
Hi, I've marked the translation to red, I can not find it in Poedit. timer would be correct but "Timer". sensor> "Sensor" ....

Another question. Why does the light turn at 06:00 when I set it from 18: 00-12: 00 I have to expand every day the time :-)
[image]

theyosh, 2017-10-25:
Ok, translation is fixed in latest version: https://github.com/theyosh/TerrariumPI/commit/3ac7a592fc70a81c074ed5ed15ef94479ce6196e

With `git pull` and clearing your browser cache should be enough to see the translation if you reload the web interface.

Time issue takes more time. I have to test it.

theyosh, 2017-10-25:
Setting time from 18:00 till 12:00 the next day? So it will be running during the night and next morning? Is this correct?

vanessa2013, 2017-10-25:
Yes correct.
Unfortunately, it does not take the whole translation, that is, large and small letters

vanessa2013, 2017-10-28:
Hi, have you seen this already?

[image]

theyosh, 2017-10-29:
That is correct! Shifting from sumer time to winter time. Tomorrow it is back to 18 hours :P

vanessa2013, 2017-10-29:
Ok, then everything works great again :-) Thank you

closed 2017-10-29 by vanessa2013