theyosh/TerrariumPI #37: Sorry next probem

vanessa2013, 2017-10-29:
[image]

theyosh, 2017-10-29:
Ok, this is explained in the release notes. Here are the needed steps:

Login in to your PI and stop the software.
`screen -r` and then press two times Ctrl+C
First update the dependencies:
`cd ~/TerrariumPI`
`sudo ./install.sh`
Now clear the old switch data:
`sqlite3 history.db "drop table switch_data;VACUUM"`
`sudo ./start.sh`

Clear your browser cache to get the latest version.

vanessa2013, 2017-10-31:
Does your program work with a DS18S20?

theyosh, 2017-11-01:
Test your DHT22 according to: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-test-a-dht11-dht22-or-am2303-sensor
This is not a problem caused by my software.

closed 2017-11-01 by vanessa2013