theyosh/TerrariumPI #38: Offline

AndrejM73, 2017-10-29:
Hi,
i installed TerrariumPi few days ago. Everything worked perfect, but after several restarts i recieved offline status and cant switch to online. Can you help me pls? thank you
[image]

theyosh, 2017-10-29:
Ok, this is explained in the release notes. Here are the needed steps:

Login in to your PI and stop the software.
`screen -r` and then press two times Ctrl+C
First update the dependencies:
`cd ~/TerrariumPI`
`sudo ./install.sh`
Now clear the old switch data:
`sqlite3 history.db "drop table switch_data;VACUUM"`
`sudo ./start.sh`

Clear your browser cache to get the latest version.

AndrejM73, 2017-10-30:
Thank you for hint, but it is still offline. I tried it wit chrome, edge and IE. Any idea whats wrong pls? Here is my log.
[LOG.txt](https://github.com/theyosh/TerrariumPI/files/1428476/LOG.txt)

theyosh, 2017-10-30:
Could you try: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

If something is not working, it will be shown there. Also, make sure you have the latest version using `git pull`

The log file is only from the installation. That does not show the problem

AndrejM73, 2017-10-30:
here is log, but i dont know what can i do with it.

pi@raspberrypi:~/TerrariumPI $ screen -r
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 40, in webserver_headers
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottle_websocket/plugin.py", line 5, in wrapper
    callback(request.environ.get('wsgi.websocket'), *args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 296, in handle_websocket
    terrariumWebserver.app.terrarium.get_environment(socket=True)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 595, in get_environment
    data = self.environment.get_average()
  File "/home/pi/TerrariumPI/terrariumEnvironment.py", line 374, in get_average
    sensor = self.sensors[sensorid]
KeyError: 'c58072c38341700885f66ce8898498be'

theyosh, 2017-10-30:
Hmmm, sounds like some issues with the configuration.

Could you rename the file settings.cfg to settings.cfg.temp and restart the software? You will have an empty system again, but that is the only way now to get it running again. If that does work, then edit the old settings.cfg.temp file in such a way that all configs regarding sensors is deleted.

These settings should start with [sensor_ID] and remove all the lines below it till the next [] part.

It should like something:

[sensor_ID]
some
settings
and
more settings

[switch_ID]
new settings etc

theyosh, 2017-10-30:
Do another `git pull` and restart with original config. It should now work, and give an error message witch sensors is causing the problem

theyosh, 2017-11-01:
No response, expect to be fixed now

closed 2017-11-01 by theyosh

vanessa2013, 2017-11-02:
[image]

Sorry, I have the same problem and all the tips so far have not helped