theyosh/TerrariumPI #39: Install help with DS1820

vanessa2013, 2017-11-01:
Hi, people can someone help me as I install the DS1820 on the raspberry, my Rasp does not recognize them.

theyosh, 2017-11-01:
Does this help? https://terrarium.theyosh.nl/static/images/documentation/ds1820-Raspberry-pi.png
Also enable 1-wire on your Pi. This can be done with: `sudo raspi-config` and go to option 5 and there P7 and select yes to enable 1wire. To be sure it is enabled, reboot your pi.

If you have wired your DS1820 correctly, it should work. Use normal pin numbering

vanessa2013, 2017-11-01:
[image]

Do you have any idea what could happen again?

theyosh, 2017-11-01:
run as normal user:
`python terrariumPI.py` in the folder TerrariumPI.

And see what it will say then

vanessa2013, 2017-11-01:
[image]

theyosh, 2017-11-01:
Strange. Looks like you have run it with a different user. But the fix is simple:

run 'sudo chown pi.pi * -R' in the folder TerrariumPI.

Then start it with 'sudo ./start.sh' again

vanessa2013, 2017-11-01:
[image]

Hmm, somehow there is a problem

vanessa2013, 2017-11-01:
[image]

theyosh, 2017-11-01:
You did it wrong! Now you have to create a fully new Raspberry pi image. You have changed the rights on the root partition and not in my software folder.

Sorry, but all these problems are not related to my software. So I will close this issue now, and advise to start completely over again. And you should try to get more understanding of Linux in general and what commands do. I cannot give support on general Linux issues.

vanessa2013, 2017-11-02:
Ok, thanks anyway

closed 2017-11-02 by vanessa2013