theyosh/TerrariumPI #40: Cannot add switches

50915091Mica, 2017-11-01:
Hi, my system collapsed and I decided to reinstall.
So there is a new lite version and fresh install. Through the webinterface I can add sensors, use the system, but the switch doesn´t save. GPIO is ok, just doesnt save..

debug shows error - terrariumswitch - pigpiod process is not running...

could it be the problem?

thanks a lot!

theyosh, 2017-11-01:
That error message can/should only be shown when you try to add a PWM dimmer switch. Any other switch should not give a problem and that error. So that is strange.

Could you try:  https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi and see what you get on screen?

50915091Mica, 2017-11-01:
Well, I managed to solve the not running pigpiod, however the problem lasts.

the debug screen mentions terrariumswitch.py, line 314 valueerror: could not convert string to float:

any sugestions? thank you

50915091Mica, 2017-11-01:
[image]

50915091Mica, 2017-11-01:
the full log is above - sorry for the way I post it, at the moment I run on a projector - improvising...

thanks in advance for your help. my iguana is getting cold:-D

theyosh, 2017-11-01:
Please update with git pull and restart it. Should work again.

commit 116d7d67 (2017-11-01): Fix #40. Keeping your iguana nicely warm
closed 2017-11-01 by theyosh