theyosh/TerrariumPI #41: Offline part 2

vanessa2013, 2017-11-02:
[image]

JoshGr33n, 2017-11-02:
I'm having the same issue. I just tried the solution you suggested at [issue#38](https://github.com/theyosh/TerrariumPI/issues/38)

pics:
[image]
[image]

Thank you

vanessa2013, 2017-11-02:
and has it helped?

theyosh, 2017-11-02:
Do a `git pull` and restart. Should be fixed

closed 2017-11-02 by theyosh

JoshGr33n, 2017-11-02:
works great, thanks!

commit f694f412 (2017-11-02): Add extra check if switch history data is available. Fix #41