theyosh/TerrariumPI #42: Request for day/night sound functionality.

arcticrobot, 2017-11-02:
Animal enrichment is very important and most tropical animals natural habitats are saturated with sounds. Sounds make animals in terrariums more active and inquisitive.

I would like to request the following functionality: 

1. Time of day. Should be adjustable by user preferences. For example: predawn 4-6, dawn 6-9, day 9-18, evening 18-22, night 22-4. Time of day and time frames should be flexible for better tuning. 
2. Sound files of natural animal habitat should be uploaded to Raspberry Pi, playlists created for each part of day and played randomly and indefinitely(until the time frame is up)
3. Volume adjustment in % (lower for evening/night, louder for day) 

I currently have this functionality achieved with Android phone and Tasker app. Sound files are stored in appropriate directories and Tasker plays them according to automation tasks set in app. Please let me know and thank you for your software.

arcticrobot, 2017-11-02:
4. Run the audio through 3.5mm jack.

theyosh, 2017-11-02:
Is it possible to upload some example audio files? Not sure what the file limits are for uploading on Github

theyosh, 2017-11-07:
I made a new branch with an audio player with all the above options. Are you able to test this? 
Also I added a web based audio player to control the volume of the current audio

theyosh, 2017-11-10:
Major sad-back: http://www.raspberry-projects.com/pi/programming-in-c/pwm/using-the-pwm-pin

It cost my days to figure out why my audio did had problems. Turns out that audio and dimming hardware does not work together on a Pi :(

theyosh, 2017-11-12:
Ok, the latest version in the feature/modules_audio branch does fully work now. It is possible to use USB soundcards instead of internal audio card.
So we can combine PWM dimmers and audio all together now.

closed 2017-11-14 by theyosh

arcticrobot, 2017-11-14:
> Closed #42 <https://github.com/theyosh/TerrariumPI/issues/42> via #47

Thank you.

I will be testing it soon. Too bad it doesn't work with dimming, I was
planning to implement that.

theyosh, 2017-11-14:
Actually it can! You need an usb soundcard. That's all.

arcticrobot, 2017-11-15:
Perfect! Thank you very much. I will be getting one, then.

PR #47 (2017-11-14): Feature/audio modules
Added audio support. Fix #42