theyosh/TerrariumPI #43: Which humid sensor should I use?

vanessa2013, 2017-11-05:
theyosh, which sensor do you use to measure the humidity?

theyosh, 2017-11-07:
I use the following hardware:
On the PI: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=30 or http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=22&products_id=73

Use for multiple sensors: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=64

For humidity: http://www.sheepwalkelectronics.co.uk/product_info.php?cPath=23&products_id=55

This is then used through OWFS and not 1-Wire self on the PI. But keep in mind, I cannot give support on how to install the hardware. If you decide to use this kind of hardware, make your self you understand it well. It is also not the cheapest hardware

vanessa2013, 2017-11-07:
Ok thx for infos. I just want something reliable.

vanessa2013, 2017-11-07:
You hava a email for Translate update.

theyosh, 2017-11-09:
Got the translation.

closed 2017-11-09 by theyosh