theyosh/TerrariumPI #44: A new problem with the interface

vanessa2013, 2017-11-06:
Since today I am no longer on my interface.
Also, not with the help of Wiki, the only thing I have changed is the alarm of the temperature from 25 to 24.5 degrees. I have again entered everything at 25 degrees, but it does not work. Can you help me?
[TerrariumPI.zip](https://github.com/theyosh/TerrariumPI/files/1446546/TerrariumPI.zip)

vanessa2013, 2017-11-06:
I could fix the problem, it was in the Settings.cfg file no name of the door contacts more registered, why always.

vanessa2013, 2017-11-06:
Why do I have duplicate entries in the settings?

[image]

vanessa2013, 2017-11-06:
[image]

theyosh, 2017-11-07:
The duplicate settings with your sensors is strange. I cannot explain. Clear browser cache? I really do not know. Your config does not show duplicate data. So that is not the problem. I have no idea.

The last error, it looks like your hardware is somewhat slow. Could you tell me how many sensors and what type they are? I do know that DHT22 can take up to 30 seconds to get the value. This is then again an indication that the hardware has problems.

If you really want to know what is happening, change the logging.conf file according to: https://github.com/theyosh/TerrariumPI/wiki/FAQ#increase-logging

vanessa2013, 2017-11-07:
Hi, I use 5x DS18B20 and 1x DHT22. Yes cache is empty.

theyosh, 2017-11-07:
An idea: get some extra pi's and sensors. So you can  experiment with one pi, and run a second one just stable. And when you know how you want your setup, you change the second one.

This is more expensive, but it will give some more freedom to experiment with different hardware

vanessa2013, 2017-11-07:
That's a good idea, I still have 2 pis at home

vanessa2013, 2017-11-08:
Hi, but I still think that it is a problem with the interface. Sometimes I can not change the period in the graphics either.

vanessa2013, 2017-11-08:
[image]

Why am I seeing the line in the chart below even though my min / max. set at 23-25 degrees?

theyosh, 2017-11-09:
Euh, please read also the help pages. There is a lot explained there if your translation is correct.

The graphs on the frontpage are averages of all sensors. So if you have 5 temperature sensors, the graphdata will be calculated based on the data of the 5 sensors. So all alarm min, alarm max and current are being averaged.

So if you alarm lines are low, it looks like you have only setup 1 temperature sensor and left the others on the default. So you have to setup all sensors. Graph updates will take 5 minutes. So you will not see it directly.

Also, the timeouts in the debug could be caused by the amount of temperature sensors. Maybe reduce the amount to a lower amount.

vanessa2013, 2017-11-09:
Hi, I think you do not understand my problem, I have 4 DS18B20 sensors connected and me all sensors are displayed twice in the interface. Of course I will not configure the sensors which are double therefore I get wrong diagrams displayed, but the problem is not 100% not on my wiring or setting but on a bug in the interface. I hope google has translated my text well. :-)

theyosh, 2017-11-09:
Ah yes. I did misunderstood you. So some questions that could help:

1. Do you see also the temperature sensors double on the page where all the senors are shown? Could you add a screen shot of that page?
2. try to open this url: http://[TERRARIUMPI]:8090/api/sensors . What do you see? Try download the page and upload here.
3. Just to be sure, add you config file again

vanessa2013, 2017-11-09:
Sorry das ist nicht möglich weil ich mein Interface wieder nicht starten kann...

[TerrariumPI.zip](https://github.com/theyosh/TerrariumPI/files/1459262/TerrariumPI.zip)

[image]

theyosh, 2017-11-09:
Sorry, but I get a bit annoyed about this. You get problem after problem. You should be able to fix those your self. Because I can never reproduce them. And there are more people using it and they do not have problems.

Also the language barrier is taken my to much time. And looking at all the issues that are created, about 80% are yours. So I have to cut back your support. You should really try to understand how Linux works. And how to debug your self. Look at the wiki pages on this project page. There is information about increasing logging which will give a lot of clues and hints. You should be able to fix any issue with that information. For now, I have to stop this issue.

vanessa2013, 2017-11-09:
Apparently something does not fit with the translation or am I mistaken

[image]

vanessa2013, 2017-11-09:
Ok, it was a missing sentence in the translation, can you please make an update, sorry.
[German Translate.zip](https://github.com/theyosh/TerrariumPI/files/1459391/German.Translate.zip)

vanessa2013, 2017-11-09:
And that comes out by the side ...
I hope this helps you further
[Sensor.zip](https://github.com/theyosh/TerrariumPI/files/1459397/Sensor.zip)

theyosh, 2017-11-10:
Ok, updated the German language. Thanks for finding the issue.

The JSON output is correct and not looking fine. It does show indeed sensors double. So could you see what the output is of the following command:

`ls -l /sys/bus/w1/devices`

theyosh, 2017-11-10:
Ok, update the latest version with `git pull`

And see if there is a difference. If it is ok now then we are done. Else you have to stop my software first and edit the settings.cfg file. Make a copy first. Then remove all the [sensorsXYZ] blocks so that there are no sensors configured any more.
And start the software. You will have to configure you sensors again, but this is the only way I think we can fix this. There is a small code change.

vanessa2013, 2017-11-10:
Hi, thanks for your help the problem is gone now. However, I had to change my Setting.cfg.

closed 2017-11-10 by vanessa2013