theyosh/TerrariumPI #45: dataplicity

50915091Mica, 2017-11-09:
Hi there,
I have more of a question than issue, I tried tunneling and controlling my PI through dataplicity, hoping that I will be able to manage the terrariumPI software from anywhere. I set up port to 80 in settings as it is the only port dataplicity is able to wormhole, but there seems to be a problem, no response, do you have a suggestion where to start if I wanted to make it work? thanks a lot...

theyosh, 2017-11-09:
Hi,

I have looked at dataplicity, blinky, ifttt etc, but all time my conclusion is that my software does the same thing. Al those sites are made to control your pi remotely. And they all have modules to read out sensors and operate GPIO. Therefore, you could accomplish the same with those sites as with my software. Read out sensors. Store the data. And build triggers etc.

So I have decided that there is no profit to made to support those platforms in my software. You can just install it and read out the sensors. But there is no way to 'upload' my software logic to those sites.

But if you want to control it over the internet, you can just use port forwarding on your router/modem. All actions that could change the behavior are behind passwords. So everybody could watch it, but only you can make changes. That is what I am doing.

So in short. The only way to open it to the internet is using port forwarding on your cable/adsl router/modem.

50915091Mica, 2017-11-10:
Thank you for explanation, I think I get it☺

theyosh, 2017-11-10:
Ok, then I close this issue. If you more questions open a new issue.

closed 2017-11-10 by theyosh