theyosh/TerrariumPI #48: Slow interface

vanessa2013, 2017-11-14:
Hi, can anyone explain why the interface loads so slowly? The cache is empty ...

vanessa2013, 2017-11-14:
And that with the door is still on my interface

theyosh, 2017-11-15:
Hmm, no sure how to debug a slow website. I do not experience it. There is 1 reason I can think off and that is that you have updated the code 24 hours ago. That means that you already have version 3.1.0. Then there is a bug that can slow down the interface. But that happens only when you visit the system status page. This bug will only running the first 24 hours after the update. Then it is gone. If you experience slow experience with version 3.0.0 I do not have a clue.

You question about the door is unclear. Could you explain it a bit more?

vanessa2013, 2017-11-15:
[image]

Hi, the interface is ok again, but what information do you need from the door?

theyosh, 2017-11-15:
Do you have configured a door sensor? Else I do not understand the problem. This is how it should work if you have configured a door sensor.

vanessa2013, 2017-11-16:
Hi, yes I have (magnetic contact). But he tells me the door was open 6 minutes ago, even though nobody was at this door.

theyosh, 2017-11-18:
Ok, this is strange. As far as I know, is there not much that can go wrong with the magnetic door sensors. Code is only a few lines and pretty stable.

So could it be that:
1. You door sensors is sensitive because the distance between the contacts is at its limit? In other words could you bring those contacts a bit closer to each other? It is about millimeters.
2. Is the sensor connected with a long cable? Is the cable extended?

vanessa2013, 2017-11-18:
My sensors are all together, and have a normal cable length, maybe there are interference from power cables ...

theyosh, 2017-11-20:
Is this issue still going on? Because I do not have any clue why this is happening

vanessa2013, 2017-11-20:
Thanks no, right now I am fighting again with my 3 DS18B20 sensors ... :-(

theyosh, 2017-11-20:
Hmmm, they all use power from the Raspberry PI. I suggest to use at least a 2Amp power adapter to power your Pi

vanessa2013, 2017-11-20:
Thanks for your tip but I already use 2.5 Amp. And the circuit is in a linear topology yet I'm always shown wrong ID's, maybe it's because of my environment because everything is in a control box, but I use a CAT6 cable for my sensors, I just want something stable :-)

theyosh, 2017-11-22:
I think I close this issue for now. I do not have a solution for this in my code.

closed 2017-11-22 by theyosh

vanessa2013, 2017-11-22:
Ok Thx.