theyosh/TerrariumPI #49: Is it possible?

vanessa2013, 2017-11-15:
Is it possible to integrate these sensors or this option into your program? I think if it works for a greenhouse too ...

https://www.amazon.de/gp/product/B071LDS4T6/ref=oh_aui_detailpage_o00_s02?ie=UTF8&psc=1

theyosh, 2017-11-15:
Hmm, sounds interesting. So I bought them and will see if I can integrate it somehow. Can take some time before the parts are in.

vanessa2013, 2017-11-16:
Cool, thanks for the support :-)

vanessa2013, 2017-11-22:
Do you have some updates?

theyosh, 2017-11-22:
Track and Trace from China :) :
最新轨迹：2017-11-20 16:30， HEATHROW / Shipment arrived at airport

So it is on its way to me

vanessa2013, 2017-11-22:
Cool

theyosh, 2017-11-24:
Ok, turns out my local electronic shop had those devices also. And I am testing it now.

But I do not think it will work well. There are two options.

- Use it digital. This means that works like a door. It is either dry or wet. There is no value of how wet it is. There is a knob to control when it switches from wet to dry or back.
- Use it analog. This means adding add an extra [HAT](https://pinout.xyz/pinout/analog_zero) to the Pi. And that involves changing pins and hoping that all hardware will still fit. Somehow it will be get a big chunky block of electronics. And that was not the purpose of this project.

So at this time, I am testing two devices. And see how I can use them. But I do have some doubts

vanessa2013, 2017-11-24:
Ok, too bad if you can not integrate it. Would fit well with a greenhouse control. But please keep me up to date.

theyosh, 2017-11-26:
I am still thinking about this..... So keep it open for now

vanessa2013, 2017-11-26:
Ok, but I do not need it any more, I will look for another software because you are doing an update after another, which does not help, but only limits my use, since the second update my sensors are not working again so there can not be any error in the hardware but to your updates, but no matter for me it is done.

closed 2017-11-26 by theyosh