theyosh/TerrariumPI #51: Is that possible part 2

vanessa2013, 2017-11-21:
I think that would be a good option to dispense with all the cables?

https://www.amazon.de/ELEGIANT-NodeMcu-ESP8266-ESP-12E-Development/dp/B018E741G4

theyosh, 2017-11-22:
I like the idea, but then I need to support 100+ devices. So that is not an option. Also this is wifi, but there is also Zigbee, XWave, and other wireless networks. That get all to expensive and difficult.

But I updated my software (beta) so that it is possible to at least read out remote temperature and humidity sensors from a second Raspberry PI running TerrariumPI. If both Pi's are on the same netwerk, you can combine them to one single TerrariumPI.

And maybe in the future more remote API's will be supported. But for now I close this issue.

A first attempt I did was using: https://getchip.com/pages/chip . This can also run Python code and read out DHTXX sensors. And has build in WiFi.

closed 2017-11-22 by theyosh

vanessa2013, 2017-11-22:
Ok Thx.