theyosh/TerrariumPI #52: I meanwhile think that it is not due to my sensors

vanessa2013, 2017-11-22:
Hi Theyosh, I'm 100% sure it's not on my sensors or on my cables. It has to be a hack on the software. Otherwise I can not explain it anymore ...

[Photos.zip](https://github.com/theyosh/TerrariumPI/files/1497443/Photos.zip)

theyosh, 2017-11-22:
Last picture indicates that the quick fix is to remove the settings.cfg file and restart. This means reconfiguring your setup. If that does not help, more investigation is needed.

I have no time to test it my self. But I do not have problems with 1wire sensors. I have two running for months now.

vanessa2013, 2017-11-22:
Should I possibly redo the whole rasp?

theyosh, 2017-11-23:
No. Only make a backup of settings.cfg and remove it from the PI. Then restart. You have to reconfigure the software again. I do not have an explanation why this is happening.

vanessa2013, 2017-11-23:
Thanks for your tip but unfortunately he brings nothing for me, currently your interface may have some errors otherwise there is not ...

[image]

[image]

theyosh, 2017-11-24:
Ah, I see.

I am sorry, but I did change the code to use only GPIO numbering. So for your DHT sensor use real GPIO port number which is 40. Enter 40 as new Address. That will also change the ID of the sensor. So after changing it to 40, go to system -> environment and chose the right sensor again and save the new environment setup.

I can a bit confusing right now. But I thought that it would be easier for the consumer to enter either only REAL GPIO port numbers or only BCM. But BCM is than limited, so I chose for REAL GPIO.

My software translate the GPIO portnumbers to right BCM numbers in code.

vanessa2013, 2017-11-24:
Hi, thanks now everything works fine. Two still ask, it is possible in the future by remote control (Nodemsu 8266) switch. And is it possible at some point to turn off the heating with door sensors when the door is open, so the heater turns off (as with the sprayer).

theyosh, 2017-11-24:
Ok, nice that it is solved.

Regarding Nodemsu: Nope. I will not make support for external devices directly. There is now an option to add remote sensors through HTTP. So technical it is possible already to support Nodemsu. But you have to program yourself the webserver on the Nodemsu. And that way, the Nodemsu could produce JSON output which can be used with my software. But again, I will not give support on how to develop for a Nodemsu.

About turning off the heater when a door is opened, I suggest to open a new issue. And stick to only 1 problem per issue.

vanessa2013, 2017-11-24:
Ok, all right, I still have a problem, my DS18B20 sensors always stay at 17 degrees, although it is warmer. It also does not help to empty Chache

theyosh, 2017-11-24:
Try this answer: https://raspberrypi.stackexchange.com/questions/14978/dallas-1-wire-temperature-sensor-not-working

vanessa2013, 2017-11-24:
Hi, the problem is not the sensor but where else, as soon as your interface queries the sensors, the sensors disappear under cd / sys / bus / w1 / devices / && ls and there is only w1 bus master present. Did you change something in the code? Only when I start Pi again, all sensors are back, until the interface interrogates again.

vanessa2013, 2017-11-25:
Unfortunately, this problem is still after the update :-)

theyosh, 2017-11-26:
Hmmmm, can you try the following:

go to /sys/bus/w1/devices

and enter a folder that starts with '28-'

there enter the following command:

cat w1_slave

What it the outcome of this? An example:

cat w1_slave   
56 01 4b 46 7f ff 0c 10 7b : crc=7b YES  
56 01 4b 46 7f ff 0c 10 7b t=21375

vanessa2013, 2017-11-26:
Yes I can

[image]

vanessa2013, 2017-11-26:
The interface does not save even if I switch from DHT22 to AM2302, despite empty Chache

theyosh, 2017-11-26:
Please one issue a time! We are now working on the temperature sensor.

So, please try to remove the DHT22 sensor from the software. So you will only have three temperature sensors left. And then there will be zero humidity sensors.
This is needed to make sure that there is no interference.

So, when the software is running only with DS18B20 sensors try to see what is happening. And then we go further.

vanessa2013, 2017-11-26:
[image]

Ok, right now there are no errors with the DS18B20, but my Pi is always offline. Should I make a new ticket for DHT22 now?

theyosh, 2017-11-26:
No, wait! We will continue on this one for now.

I see an error getting the weather data. Have you updated your weather settings? Or could you check what is you current weather data source?

vanessa2013, 2017-11-26:
I always use this link:
https://www.yr.no/place/Austria/Vienna/Simmering/

theyosh, 2017-11-26:
Ok, could you restart your pi? The weather source is working for me.

vanessa2013, 2017-11-26:
[image]

Ok it is currently running, then it may be the DHT22 setting

theyosh, 2017-11-26:
Ok, so keep this running for let say 10 minutes. To make sure it is working without errors.

Then we close this issue and open a new one for the DHT22 problems. Ok?

vanessa2013, 2017-11-26:
Ok, thank you very much

theyosh, 2017-11-26:
No, still wait!. You add new 'problem' again. :)

What is the door name you have entered? Because the first pictures shows an error while saving the door configuration.

vanessa2013, 2017-11-26:
Upps Sorry, I took the name intermediate door.

vanessa2013, 2017-11-26:
Zwischentüre

vanessa2013, 2017-11-26:
Maybe because of the ü

theyosh, 2017-11-26:
yesh. That is a problem....

theyosh, 2017-11-26:
Could you try it first using a normal 'u' ?

vanessa2013, 2017-11-26:
Ok then I will fix that, I have a question for an extension sorry but I have a few ideas to say simply if it bothers: -...

theyosh, 2017-11-26:
Euh.

Last step for now is adding your DHT sensor. Keep in mind you have to use physical pin numbering. So count it according to https://pinout.xyz

vanessa2013, 2017-11-26:
Ok what should I select DHT22 or AM2302?

vanessa2013, 2017-11-26:
I have this: https://electronics.semaf.at/DHT22-Temperature-Humidity-Sensor

theyosh, 2017-11-26:
Hmmm, that I would not know. Try AM2303 first. If that does not work, change it to DHT22. But make sure you test every time for 2 a 3 minutes. If there are problems it will take some time to show the error.s

vanessa2013, 2017-11-26:
Ok, I have to restart because so the error under screen -r go away?

theyosh, 2017-11-26:
which error?

vanessa2013, 2017-11-26:
[image]

vanessa2013, 2017-11-26:
If everything fits, but only the first two lines to see

theyosh, 2017-11-26:
That problem is due to the German language. Could you edit the settings.cfg file and change all the umlauts and other typical German letters to normal letters? That will fix the above problem

vanessa2013, 2017-11-26:
So "I have to change the German translation completely :-)

theyosh, 2017-11-26:
No! Only in your settings.cfg file. The translation is just fine

vanessa2013, 2017-11-26:
Now I have reloaded Putty and get that as a message.
[image]

theyosh, 2017-11-26:
above error is because you have on a different putty the software already running. Please reboot

vanessa2013, 2017-11-26:
Oh yes sorry my mistake, but as I said my setting file is written normally and the errors remain inside.
I have now done a complete reboot now comes this:
[image]

vanessa2013, 2017-11-26:
Oh man I hate this fucking DHT22 sensors
[image]

theyosh, 2017-11-26:
Why is you Pi having network problems? Again it cannot retrieve weather data. The second line is the problem. So I have added more code to fix this issue. Please update the code with:

`git pull`

and restart

vanessa2013, 2017-11-26:
Ok the weather problem is settled, but I have problems with the DHT22 again :-(

[image]

theyosh, 2017-11-26:
What pin number have you entered?

vanessa2013, 2017-11-26:
[image]

theyosh, 2017-11-26:
Still, the question is. What number have YOU entered?

vanessa2013, 2017-11-26:
[image]

vanessa2013, 2017-11-26:
[image]

There are also my sensors gone.

theyosh, 2017-11-26:
Ok, that is correct.

Then, please remove the DHT sensor here by clicking the small x on the left and add a new one with the type DHT22. Do not just change it in the settings screen. Make sure you have removed the old one and added a new one.

Your temperature sensors are gone due to probably failing DHT sensor... but please ignore for now. The problem is your DHT sensor

vanessa2013, 2017-11-26:
Ok, even with the setting DHT22 is not since the last update is so

[image]

vanessa2013, 2017-11-26:
Can it be that a 400 volt line disturbs the sensors

theyosh, 2017-11-26:
Sorry... my fault... will make fix first

vanessa2013, 2017-11-26:
No problem just let me know, sorry if I'm so annoying

theyosh, 2017-11-26:
do a `git pull` and restart again. Pleas do not ask multiple questions. Yes, 400 volt power lines will influence your data lines....

But please add just ONE DHT22 sensor to test with. I suggest to use it as a humidity sensor. The more sensors you add... to more complicated it makes. And that is not helping due to the fact I have to ask everytime what you have changed.... This takes to much time. So 1 small step a time please!

vanessa2013, 2017-11-26:
Ok, after your update it's like this again ...

[image]

vanessa2013, 2017-11-26:
Without DHT22 everything fits perfectly

theyosh, 2017-11-26:
Ok, at this point I think you should able to figure out that a sensor is missing. So update your environment. 

But I stop for tonight. This takes to much time. You know how to test a DHT22 sensor without my software. I have explained that multiple times.

So, for now, I still after days off testing and trying, it is your hardware that is failing. And that I cannot fix. So I conclude that my software is not the problem.

Please do proper testing with your hardware and make sure they work before asking questions.

[3 more replies]

closed 2017-11-26 by vanessa2013