theyosh/TerrariumPI #54: Switch off heating with door control

vanessa2013, 2017-11-24:
Is it possible to switch off the heating with the door sensor when the door is open and when the door is closed again switch it on again (as with the sprayer).

theyosh, 2017-11-26:
This is still open to decide...

theyosh, 2017-11-26:
The problem with this is that I have to keep track of the heater state. And that is a bit difficult.

The sprayers does also work a bit different. The sprayer will not START to spray when the door is open. But when the sprayer is started, it will not stop when the door is opened. So it will not interrupt the sprayer when actual spraying.

So, what is possible, is to make an option, that the heater will shutdown when you open the door, and that the heater stays out when the door closes again. Due to temperature drop in time, the heater will start when it gets below the min alarm. So that is an option.

theyosh, 2017-11-26:
No need anymore

closed 2017-11-26 by theyosh