theyosh/TerrariumPI #55: translation issues

vanessa2013, 2017-11-24:
Unfortunately, I am unable to translate everything.
[image]

theyosh, 2017-11-25:
I have added German translation for the audio file listing.

But, I have also introduced 5 new sentences. The are in the dropzone part. If you could translate those as well, that would be very nice

vanessa2013, 2017-11-25:
Hi, ok here are the files, but everything is still not translated ...

[Translate.zip](https://github.com/theyosh/TerrariumPI/files/1503394/Translate.zip)

theyosh, 2017-11-25:
What part is now still not translated? I do know that not all parts of the weather page is translated. But that is impossible due to external terminology. So I cannot translate that.

vanessa2013, 2017-11-25:
The picture I sent you yesterday ...

[image]

theyosh, 2017-11-25:
After latest pull, clear browser cache. This translation is done client site in the browser

vanessa2013, 2017-11-25:
Nop sorry, 

[image]

theyosh, 2017-11-25:
Hmm, turns out that the German translation is not valid JSON:

https://cdn.datatables.net/plug-ins/1.10.16/i18n/German.json

So this is also a problem I cannot fix. It is a bug in an external plugin. https://datatables.net/

vanessa2013, 2017-11-25:
Ok you can not do anything, then this topic is done :-)

closed 2017-11-25 by vanessa2013