theyosh/TerrariumPI #57: Question for extending the system environment

vanessa2013, 2017-11-26:
Hi Theyosh, again a "suggestion or idea".
Is it possible to customize the system environment by the user, for example: not only sprayers, light, heating, air conditioning but also other things like e.g. Fan 1, fan 2, or something else. As a box with which you can design your own system environment, with options for what are currently possible (with light disable, with door open disable, etc. Because currently the user has no possibility to set his own system environment. Do you understand what I mean ( unfortunately the translation is not good).

theyosh, 2017-11-26:
I have thought about this, but I will not make this. Then it will lose the purpose which is running a reptile environment.

I do understand what you want, but there are other parties that do this.

https://ifttt.com/
https://www.dataplicity.com/
https://thingspeak.com/

etc. So there are a lot or other ways to make a very dynamic setup. But I will not make it in my software.

vanessa2013, 2017-11-26:
With these programs, I can do the same as with your just stop more I have to adjust everything myself or how?

theyosh, 2017-11-26:
Sorry, I am not going to explain what other people are making.

I will not make what you have asked here. I offered you different options. I suggest you go look at those sites to see if it helps you. It is not up to me to explain other people systems or software.

closed 2017-11-26 by theyosh