theyosh/TerrariumPI #60: Relay Switch dont stay "ON"

knuckyball, 2017-11-26:
Hello,
First a BIG thanx for all your work.
I just began to use it (or..."try" to be exact) yesterday.
The sensors are working, the switch is configured and working when i push manually on the interface button. But after a few seconds, it switch OFF ;-(
And i configured the environment to switch ON heater with sensors...and nothing happend...when the low temp is reached, the switch dont turn ON.
Any ideas to help a newbie ? ;-)

knuckyball, 2017-11-26:
I understand what you mean, but it's not the same situation here.
The Switch is for the heat lamp (for a pogona) and the conditions are linked to the temperature sensor.
I've configured the alarm to 23 degrees C and the sensor is at 21 degrees. The alarm state is triggered but the Switch dont turn on. And when i trigger it manually, it Switch off after some seconds.

vanessa2013, 2017-11-26:
Ok then you have to wait for @theyosh Theyosh who knows better

theyosh, 2017-11-26:
Ah, This is because the heater is designed to run when the lights are off. For my case, the lights give enough heat so I do not need extra heating. So the heater normally works only when the lights are off.

To change this, go the the environment page, and enable the option 'enable when lights are on'.

That should make the heater work 24/7 a day. Even when the lights are on and producing heat

theyosh, 2017-11-26:
Also, I disabled the dependency between the lights and heater in code

vanessa2013, 2017-11-26:
What exactly do you mean by that?

theyosh, 2017-11-26:
That the heater would not work if you did not enable the lights system. But I can imagine a setup where there are no lights needed, but still need to spray or heat it. That is now possible also.

vanessa2013, 2017-11-26:
Do I understand this correctly when I want to heat I also have to turn on the light? Sorry for the stupid question but using google Translate is always so weird

theyosh, 2017-11-26:
Lights need to be configured / setup. That is all.

vanessa2013, 2017-11-26:
Ok thx.

knuckyball, 2017-11-26:
Thanks a lot.
I'll try as soon as i get back to Home ;-)

knuckyball, 2017-11-26:
GREAT ! I just tested with light configured and all is OK. The switch for heat is working properly !
A big big thanks !

closed 2017-11-26 by knuckyball