theyosh/TerrariumPI #61: Problem since update

vanessa2013, 2017-11-26:
[image]

theyosh, 2017-11-26:
Your pi is not able to resolve DNS. Please restart your PI

vanessa2013, 2017-11-26:
[image]

I do not understand it my sensors always show correct values on the Pi only in the interface

vanessa2013, 2017-11-26:
Your interface has a lot of problems with DHT22 and DS18B20

vanessa2013, 2017-11-26:
Can it be on my port?

vanessa2013, 2017-11-26:
[image]

theyosh, 2017-11-26:
Close this. Same as https://github.com/theyosh/TerrariumPI/issues/52

closed 2017-11-26 by theyosh