theyosh/TerrariumPI #62: RPIcam change

stevoace, 2017-12-03:
After initial setup of a RPICam the pi was powered down and camera replaced by an alternate IR Cam, at this point the webservice never started. I returned the initial cammera to the pi and stilll no webservice start. after SSH connection to the Pi no screen was active. After starting (./start.sh) i watched the screen and recieved the following output:

 02:53:46,000 - INFO - terrariumWrapper - Starting TerrariumPI server ...
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
    terrariumEngine = terrariumEngine()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 86, in __init__
    self.__load_webcams()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 199, in __load_webcams
    webcams_config[webcam_id]['rotation']
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 60, in __init__
    self.update()
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 303, in update
    self.__get_raw_image()
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 69, in __get_raw_image
    stream = self.__get_raw_image_rpicam()
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 133, in __get_raw_image_rpicam
    logging.error('Error getting raw RPI image from webcam \'%s\' with error message: %s' % (self.get_name(), err,))

I resolved this issue by removing the webcam entry in the settings.cfg and re-adding the cam.

theyosh, 2017-12-04:
So to be clear. First attempt with normal/original cam did work.
Changing to a different cam give errors?
Changing back again to original did not work until you cleared the settings?

I do know, that some errors are not log-able, and produce a second error. But at the moment it is working?

theyosh, 2017-12-10:
Closing and see this as informative

closed 2017-12-10 by theyosh