theyosh/TerrariumPI #64: Cosmetic issue: Uptime on dashboard

Aapjes94, 2017-12-12:
I'm still setting everything up at the moment but noticed a quirk. The System Status bars overlap the uptime counter.

[image] 

It is a small cosmetic issue, which probably isn't even noticeable on a wide screen.

theyosh, 2017-12-12:
Hmm, actual this is by design :P . I thought that the bars should go over the uptime. Because the last seconds are not that important. Else I have either to remove the cpu usage bars, make smaller font etc.

And CSS is not my strongest part in coding. And yes, I do have a wide screen :)

But what would you expect? Because for me it is not an issue. So I have a bit of a hard time to figure out what a good solution would be.

Aapjes94, 2017-12-12:
I am not really experienced in CSS either, so I cannot really offer an alternative. I guess I shoud consider getting a wider screen :) 

Feel free to close the issue.

theyosh, 2017-12-12:
goed, dan sluit ik de issue ;)

closed 2017-12-12 by theyosh