theyosh/TerrariumPI #66: Cant add sensor

badner, 2017-12-28:
Hey, I really like your project and was able to get everything done so far. But now I'm stuck at adding a am2302 temp/hum sensor. I wired everything like explained in your program. But somehow it doesn't want to recognize the sensor. Would be nice if you could held. 
Thanks

theyosh, 2017-12-31:
Have you tried: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-test-a-dht11-dht22-or-am2303-sensor

Pay attention to the pin numbering. For testing with the link above, use BCM numbering. See for translation the pinout.xyz page. When the tests are working, you know the hardware is ok. Then you should be able to add it to my software.

Also you can try:
https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Barbara1984, 2017-12-31:
Can it be that there are problems with the Pi3 Bitbang when connecting two DHT22?

theyosh, 2018-01-01:
Not that I know off. But I can't judge without debug logging. So try debugging according the links in the first message. And post outcome here.

theyosh, 2018-01-01:
Make sure you use different GPIO pins for each DHT sensor. They can't share GPIO

badner, 2018-01-02:
U Legend!

closed 2018-01-02 by badner