theyosh/TerrariumPI #67: Sprayer mode

Barbara1984, 2017-12-28:
Hi Theyosh, thanks for your great project !!
One question, is it possible to integrate a dehumidifier into your program? Or maybe a Max. Function with Sprayer mode?

theyosh, 2017-12-31:
Hi,

Euh, the sprayer should only spray when the humidity is to low. So then there is no need for a dehumidifier. The only need for a dehumidifier is when the humidity is always to high. But then I use more heating to get it lower.

The spraying system use 1 or more humidity sensors. With those sensors you set the min and max values. If the max value is close to the min value, you are able to reduce the spraying time to a few seconds. So you can spray small amounts.

But if you need something to get a lower humidity when the lights are off, then it is not possible at this time. I am using a humidity value that is the same all day. And a high humidity during the night is good for my animals.

Barbara1984, 2017-12-31:
Hi, thank you for your answer, will this function maybe come sometime? And is it somehow possible to make a time switch function because of switches that I could control more about time? if you know what I mean (just insert a small box with from-to at each switch)

theyosh, 2018-01-02:
Euh, now it gets confusing.

What I can do I change the humidity system in such a way that it can control a device for spraying when it is to dry. And run a dehumidifier when it is to high. But that can turn in to loop between spraying and dehumidifying. So not sure if that is something that will work.

Spraying on a predefined period is a bit harder to do.

So could you explain a bit more what you are looking for. I suggested multiple options.

Barbara1984, 2018-01-03:
Hi, I need a function to all devices synonymous per time individually to controll. So I imagine that at each switch a field with time is inserted where you can enter the time how the device should run. I also have devices that do not fit into the system environment and I would have to control them individually. Do you know what I mean?

Barbara1984, 2018-01-03:
So in the way as with the switches, Gpio, Gpio reversed, dimmer, and then possibly "timer" ...

theyosh, 2018-01-06:
Hi,

I understand your request. But I am having a bit of an issue with that. Because this means that every switch needs its own timer routine. Which is more then just a value for a duration. It needs to keep track of previous states in order to determine when to go on and off. Also this brings a lot more internal processing with it.

So I think I have to say that this request is not something I am going to make. For me there is no use at all. And as I said. You need more information then just that it is a timer. Need start and stop times if it should not run all day. Then we also need two duration values for on and off.

So, I have to think a bit more if this is something I would add this to the software.

Barbara1984, 2018-01-06:
Hi, ok thanks, just let me know.

theyosh, 2018-01-10:
I have created a new feature for tracking #72

closed 2018-01-10 by theyosh

Barbara1984, 2018-01-10:
Yes, thx