theyosh/TerrariumPI #68: Questions about alarm usage

jprjr, 2017-12-30:
I had a question about alarms: it looks like I can set a minimum and maximum alarm temperature, I assume the alarm is triggered when it goes below the minimum, or above the maximum.

When an alarm is triggered, what happens? Is that what the audio files are for?

theyosh, 2017-12-31:
When an alarm is triggered, it will power on or off the switches that are defined to the sensors in the environment page. For example, if the humidity sensors detect a to low humidity, it can power on the switch which holds a sprayer system. Same goes for heating during the night.

When no switches are configured, the alarm is only visual.

The audio files is a different module of my software. It is only able to play sounds through a playlist. Is not used for alarm.

jprjr, 2017-12-31:
Sweet - thanks for the clarification!

closed 2017-12-31 by jprjr