theyosh/TerrariumPI #69: Feature Request: CSV / XLS export

jprjr, 2017-12-31:
AFAICT, sensor data, weather data, etc is stored in a Sqlite database.

It would be really convenient to be able to download CSV/XLSX data via the web front-end.

theyosh, 2018-01-01:
I agree. So a CSV export will be added.

theyosh, 2018-01-02:
If you check out the new code, there will be an export option in the settings menu in each graph

Aapjes94, 2018-01-04:
After the update for this feature I've been getting these errors:

2018-01-04 16:01:56,019 - ERROR - terrariumCollector - TerrariumPI Collecter exception! (DatabaseError('database disk image is malformed',),)

2018-01-04 16:01:56,024 - WARNING - terrariumCollector - TerrariumPI Collecter recovery mode is starting! (True,)

Is this related to this update or have I messed up somewhere else?

theyosh, 2018-01-04:
No, this is the result of my bad programming. I did not make a clean shutdown. So it could be that you terminated the software while writing to sqlite. But the recovery should take care of it.

If not, the only solution is to delete the history.db file. It will then be recreated with a new restart

theyosh, 2018-01-10:
So, for now I close this. Export is rather simple, but effective for now

closed 2018-01-10 by theyosh