theyosh/TerrariumPI #70: build log

scrumpydc, 2018-01-08:
is there any chance of a build log or something to help out noobs to raspberry pi or is this something you will not be doing ie were to attach things on the raspberry and how to set up ive been trying the raspiviv but it looks like he is not supporting it anymore and this may work for me but have no idea were to go or is this something that you would not recommend for someone fairly new to raspberry i have relays and dht22 sensors 
also are you able to set times for each individual relay ??

Jelly-Flinger, 2018-01-09:
Hi, 

I am noob also. I have been trying to set up the system and making reasonable progress. I looked at raspviv  too but decided this might work better for me.

As a noob I would start with your dht22 sensors. When I added them I thought I had wired them correctly but could not seem to add them in the sensor settings page.

I used this https://zabbix.tips/monitor-temperature-and-humidity-zabbix/ tutorial to get started (up to and including  the test your sensor section). However, I later found a similar procedure to test in the FAQ. 

Turns out I had wired the sensors correctly. When I ran in screen session (also in FAQ section) I could see some errors. I had been assuming I only had to enter the fields marked with an asterix so I was ignoring alarm and limit. When I populated all the fields I could add the sensors and they started showing up as part of the sensors/settings page and also the average temperature on the control panel. 

The only question I have is, why can I not see trends in the temperature and humidity page? I originally hooked up some DS18B20 temperature probes which autodetected and showed up as trends (I think). 

I also have question about the individual switching of lights. I plan on controlling the heat lamp from a dimmer eventually but will use on/off to begin with. Looks like there is an entry for that.

The UV and basking light I want to control individually. I figure I can use the lights  entry for one and the mister for the other (I don't plan on using an actual mister). Failing that I guess it's put in a feature request or try and adapt the code for yourself. 

I am going to try and wire up the relays next so will feed back on anything that may be of help to a fellow noob!

Aapjes94, 2018-01-09:
This was basically my first raspberry pi project, and I've managed to get this all up and running without too many problems. Feel free to ask me if you're getting stuck, and I'll see if I can help. 

With the equipment you have, you should be able to get a working basic version up and running in a few hours.

> I figure I can use the lights entry for one and the mister for the other
This an indeed an issue, the environment page could use some expansion.

theyosh, 2018-01-10:
Ok, this software is made with the idea that you should have some basic knowledge with a Raspberry pi and even hardware sensors. And therefore, I do not provide a detailed howto's. Sorry for that, but it is still a hobby project for me that I do in my free time.

So it depends what is needed. To start with, my software does NOT support timed relays. It has been asked by other people also. So it looks like there is some need for them. I need some time to think about it how to implement it in a way that everybody is happy.

Secondly, I have used physical GPIO pinout for all sensors and switches / relays. So the DHT22 normally use BCM GPIO layout, I use the physical layout. So if the DHT22 is working ( https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-test-a-dht11-dht22-or-am2303-sensor ) you test it with BCM numbering. Then count it and use the physical number in my software.

In general, is you have worked with DHTxx sensors in the past, this is not much more complicated. As long as you get a working Raspberry pi and follow my installation steps ( https://github.com/theyosh/TerrariumPI#installation )

I think you mean the graphs when speaking about trends? The graphs take some time before they generate. At least 5 minutes. And due to caching it normally update in the browser once every 5 minutes. The data is measured every minute. So clear browser cache and reload the page?

For now the dimming support is somewhat limited I think. You can configure a dimmer switch with the min and max amount of power to use. And the time it needs to take to go from min to max or vice versa. And I have not tested a dim duration of more then 90 seconds.
But do I understand that the heat lamp is using different timing then your UV lamp? In my case they have the same time schema. If that is also your case, you can combine those lights together in the lights settings on the environment page.

I hope this helps a bit. This is also the limit I can go. But if you are able to login with SSH on a PI, and can confirm that the sensors are working, I can help out anytime. That is enough to get this all running!

closed 2018-01-13 by theyosh