theyosh/TerrariumPI #72: Timer functionality per switch

theyosh, 2018-01-10:
Add an option to created separated timers for switches. This will allow the switch to turn off and on on a time based schedule.

Needed input per switch:

- Timer start time
- Timer stop time
- Timer on duration
- Timer off duration

In which the duration between start and stop is 10 hours and the on duration is 1 hour and the off duration is 2 hours, it will be on for 4 hours in total.

If this timer is enabled on the switch, it cannot be used on the environment page. Because the switch has its own timer logic and not using the central logic anymore

Aapjes94, 2018-01-10:
I tried to update but things got messed up. From the TerrariumPi folder, I ran sudo git pull, but got the message that it was already up to date. Not sure if what I did next was stupid but I ran ./install anyway and now corrupted the whole thing. 

Can you confirm that it was stupidity on my part? Or did something strange happen due to the code?

theyosh, 2018-01-10:
Please reopen new issue. Your problem is not related to this issue

Barbara1984, 2018-01-10:
Nothing has been uploaded in the code here :-)

Barbara1984, 2018-01-22:
Hi, I do not want to be silly but when will this update come about? :-)

theyosh, 2018-01-25:
Updates for this feature can be tested in branch https://github.com/theyosh/TerrariumPI/tree/feature/switch_timers. So update with `git pull`. Then use `git checkout feature/switch_timers`. Then restart TerrariumPI and clean your browser cache.

Also keep a copy of your settings files before testing this.

Barbara1984, 2018-01-25:
Ok, thanks, I'll tell you how it works

Aapjes94, 2018-01-25:
From my end, this branch seems to work. The update worked and it's installed. The 'translations' are still missing, so it isn't exactly clear how this function work. Especially the interactions with the environment settings.  I'd assume timed switches won't be controllable with the environment settings, which wouldn't make much sense.

theyosh, 2018-01-27:
At this point I think I got it working. So it could be fully tested now.

Barbara1984, 2018-01-27:
Very cool thanks for your work I will test it tomorrow.

theyosh, 2018-01-28:
Clear browser cache, and select active at the timer pull down when editing the switches.

Barbara1984, 2018-01-28:
Sorry maybe I'm too stupid but I think nothing has done the update and emptied cache. But I can not find the timer mode.

Aapjes94, 2018-01-28:
It worked for me so something went wrong during your install. Are you sure you used `git checkout feature/switch_timers` after `git pull`

`cd TerrariumPi`
`git pull`
`git checkout feature/switch_timers`
`sudo ./install.sh`
probably a `sudo reboot` is needed after this as well to get everything set up properly

Barbara1984, 2018-01-29:
Thanks for the help now it works :-)

theyosh, 2018-01-29:
Cool, I have it also merged to master branch. So everybody can use it now :+1:

Barbara1984, 2018-01-29:
Can you please # 82 again open the subject is unfortunately not done :-(

theyosh, 2018-02-03:
Update the code that timer functionality is also in all the environment parts. Same logic.

closed 2018-02-08 by theyosh