theyosh/TerrariumPI #76: Trouble with fresh install: No module named uptime

Aapjes94, 2018-01-11:
Started with a formatted SD card and did a fresh install. Followed the install instructions in the FAQ, but ran into some problems. 

1.  Screen wasn't installed with the rest of TerrariumPi
2. I got some gevent error, which I soved by first installing pip and then gevent
3. Now I'm stuck at a new error "No module named uptime"

`2018-01-12 00:23:27,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.178.24:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 11, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 7, in <module>
    import uptime
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 93, in __import__
    result = _import(*args, **kwargs)
ImportError: No module named uptime
2018-01-12 00:23:28,000 - INFO - terrariumWrapper - Restart counter: 1/5`

I've tried updates and upgrades and reboots as well.

theyosh, 2018-01-13:
Hi, 

this all sounds strange. Did you run the installer script first? Because the installer script will install screen and will install the python uptime modules.

So please run the install.sh script from inside the TerrariumPI software folder

Aapjes94, 2018-01-13:
Hi,

It is indeed a bit strange. I did yet another reformat and had the same issue. Now I ran the install script a second time and it worked. I'm 90% sure I ran if from within the TerrariumPi folder, but cannot guarantee it. I've got it up and running now though. 

I've made a disk image of everything now, so I shouldn't have to get these issues again. If something happens I'll make sure to do the install from within the TPi folder and report back. 

I'll close the issue now.

closed 2018-01-13 by Aapjes94