theyosh/TerrariumPI #79: AM2302 readout in dashbord

vankomicin, 2018-01-15:
Hi,
in terminal I can correctly read the AM2302 with
sudo ./AdafruitDHT.py 2302 25
but when I add the sensor in web interface, there is no readout, it also doesn't show the humidity graph.
Any ideas?
[image]

theyosh, 2018-01-15:
Ah, that should be easy to fix:

https://github.com/theyosh/TerrariumPI#gpio-numbering

So you have tested number BCM 25, which is 22 in Physical pin numbering. So changing to 22 should fix it. You have to that for both types for the AM2302 (Temp and Humidity).

I will also update the FAQ about this

vankomicin, 2018-01-16:
Thanks! Now it works. Should have read the FAQs really carefully because this physical and BCM relationship in software. One thing I noted however is the latency before the readings show up in web interface...
Keep up the good work!
Rok

closed 2018-01-16 by vankomicin