theyosh/TerrariumPI #80: Total power (kWh, L) and total price

Barbara1984, 2018-01-20:
Hi, is it somehow possible to reset the overall performance + total price and get it back to zero?

Aapjes94, 2018-01-21:
That data seems to be logged in the history.db file. The contents of that file looks like gibberish from Terminal, so I couldn't tell you what that file looks like at a fresh install. If you could find that out you could replace your current file with a 'blanc' history.db file.

theyosh, 2018-01-22:
The history.db is a sqlite3 file. So use sqlite3 and sql command to read the history.db file if needed.

But the fastest way to 'reset' the data is:

- stop the TerrariumPI software ( https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-stop-terrariumpi )
- remove the history.db file ( rm history.db )
- start TerrariumPI software ( https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-start-terrariumpi 
- )

This will recreate the history.db file with empty values. This will also delete all the existing graph data of the sensors and power switches.

Barbara1984, 2018-01-22:
Thanks, theyosh, I'll do it that way.

closed 2018-01-22 by Barbara1984