theyosh/TerrariumPI #81: Not everything is installed when running the ./install file

Aapjes94, 2018-01-21:
After having installed TerrariumPi a couple of times I've noticed an error pattern in the installation. It appears that line 32 of the install is not running properly. 

`pip install --upgrade gevent untangle uptime bottle bottle_websocket pylibftdi pyalsaaudio`

After following the installation instructions it won't run after a reboot. When manually running ./start.sh it starts a restart cycle because all the files which should be installed in the above line appear to be missing. 

Running the ./install.sh again after a reboot installs everything properly.

theyosh, 2018-01-22:
Hmm, this I have to investigate. But it will take some time because I have to create empty Raspberry PI images.

But I will look into it

Aapjes94, 2018-01-22:
Thanks.

The workaround is of course really easy, but I just thought I'd let you know. When you decide to look into this, let me know if I can provide any more information.

theyosh, 2018-02-03:
Just did a test and no issues. So I think it was a hickup. Also pip install needs to download modules. Maybe that didn't work at the moment you tried to install.

I will close this ticket, due to probably a hickup and not a recurring bug. Thanks for sharing

closed 2018-02-03 by theyosh

Aapjes94, 2018-02-05:
I still had the same issue when testing 2 days ago. If you're not able to reproduce the issue, just leave it at this. If I'm the only one experiencing this, it's probably related to my set-up. Feel free to leave it closed. Thanks for looking into it though.

commit 4a1ef7a3 (2018-02-11): Changed installation script to support other user then pi to run the software #96. Changed pip installer so it could fix #81. And added support for remote usb power switches #95