theyosh/TerrariumPI #82: Universal AC MAINS Dimmer - MPDMv4.1

Barbara1984, 2018-01-21:
Can someone help me with the installation, I do not bring it together that I can dimming a fan.

Aapjes94, 2018-01-21:
What type of fan do you have and with what settings did you try to add it? Also important is what GPIO pin you connected it to.

Barbara1984, 2018-01-21:
Hello, it's a 230V toilet fan and it's connected to the GPIO for PWM GPIO18. But unfortunately it does not work I think I made a mistake during installation, but I do not know 100%.

Aapjes94, 2018-01-21:
PWM on the RPi 3 is only available on GPIO 12, 32 and 33, as can be seen on this [website](https://pinout.xyz/pinout/pin32_gpio12) The different numbering systems can be a bit confusing, but using the GPIO numbering used in TerrariumPi is appears you're using the wrong GPIO pin. 

For testing you could use the same settings and pin as TheYosh as can be found [here](https://terrarium.theyosh.nl/).

Aapjes94, 2018-01-21:
I just realised something. Unless you have a special type of fan I doubt it will work, as I can't imagine a 230V bathroom fan is PWM dimmable. For that you usually need a fan with 3 wires. A set for the positive and negative leads as well as a signal wire. 

As you most likely use 12V somewhere else in your setup, it will be easier to find a PWM dimmable 12V computer case fan. I got a set of 120mm fans for less than €7.50 each. Cheap, powerfull and very silent as well.

theyosh, 2018-01-22:
For PWM dimming, there are only  GPIO 12, 32 and 33. So make sure you are using one of those.

Further, it depends on the fan, if it will work with a dimmer. It will just get less power. So the Fan does not have support for PWM. You will have to have a dimmer module like: https://www.tindie.com/products/nEXT_EVO1/universal-ac-mains-dimmer-mpdmv41/

So the module will dimm the power from 0 to 250Volts. And if your fan can handle that, it should work.
I use this module with a 220V bulb light.

Barbara1984, 2018-01-22:
Hi theyosh I use exactly this board as you indicated in the link, I'll try it tomorrow with another pin

Barbara1984, 2018-01-27:
Hi, so I've tried it now, unfortunately without success as you have this board installed (software and hardware) I have now connected like Theyosh. Many thanks for your help!

theyosh, 2018-01-28:
Ok another way of solving this:

- Update my code with git pull on master branch
- Run the installer. Sudo ./install.sh . This can/will update packages or extra tools that are needed
- Restart the PI
- Make sure the program pigpiod is running
- Test it with a normal 220/250V light bulb.

If this does not change anything, I do not know what it could be.

Barbara1984, 2018-01-28:
Do I just have to install Pigpio and nothing else, right?

theyosh, 2018-01-29:
just run the install.sh script. It will take care of it

Barbara1984, 2018-01-29:
Okay I did that (today after the "update").
Unfortunately that did not help. So he is connected or configured.

[image]
VCNT = GPIO 32
+VCC = 3,3V
GND = ground

[image]

theyosh, 2018-01-29:
Hmm, this looks ok. As long as the small led on the board is lighting up, it should be connected ok.

Are you still testing with the fan? Because it could be that the fan has a 12v transformer and that can't be dimmed. So I always test with a normal light bulb.

So if you have checked that pigpiod is running, test it with a normal light bulb. Also make sure that you use physical pin 32: https://pinout.xyz/pinout/pin32_gpio12#

Finally, you can use this piece of code to test it manually. The number 12 is correct here. Create a python file and run it. It will ask for some numbers which will change the dimming. 

`import pigpio
GPIO=12

pi = pigpio.pi()
if pi.connected:
   print("Connected to pigpio daemon !\n")

dm = 0
while dm < 860:
    dm = input("Enter Dimmer Value: ")
    dm = int(dm)
    pi.hardware_PWM(GPIO, 5000, dm*1000) # 5000Hz dm*1000% dutycycle

print("\nOut of range. Exiting...")
pi.hardware_PWM(GPIO, 5000, 860000) # 5000Hz 95% dutycycle
pi.stop()`

Else, you could try the creator: https://www.tindie.com/products/nEXT_EVO1/universal-ac-mains-dimmer-mpdmv41/

Barbara1984, 2018-01-30:
Ok, thank you, I'll test tomorrow, I'll let you know

theyosh, 2018-02-03:
Any updates? I did notice that the TCP connection can be an issue in code. If so, there is a small fix for that. But please confirm first that the tests works.

Barbara1984, 2018-02-03:
Hello, so somehow it will not work for me :-(

theyosh, 2018-02-03:
Ok, then I am afraid that I cannot help further. As I am not the maker of the board, I also have to Google for answers. 

So a final option that I can think of, is to complete install a new clean Raspberry PI. Do NOT install my software or anything else. Just a clean Pi. Then try to get your dimmer working by the following his youtube movies. http://www.esp8266-projects.com/2017/04/raspberry-pi-domoticz-ac-dimmer-part-1/

Sorry that I cannot help you further with this.

Barbara1984, 2018-02-06:
I hope I come to this week to test it.

theyosh, 2018-02-11:
How do did you switch it off? Used the slider and set it to zero percentage ?

And did you set the off dimmer value to 0% in the power switch settings?

Barbara1984, 2018-02-11:
Currently my values look like this:
[image]

Yes, I turned the switch to 0% and it's blue when it's off and green when it's on.

Barbara1984, 2018-02-11:
You mean that value, I'll change it now and let me know.
[image]

theyosh, 2018-02-15:
Did you already found the right value? If higher does not work, you could even try to go as low as 800 to see what happens.

Barbara1984, 2018-02-18:
Hi, thanks tomorrow morning I can tell you more :-)

Barbara1984, 2018-02-22:
So the lamp glows a bit what should I do now?

Barbara1984, 2018-02-22:
So 895 is the right value so that the lamp no longer glows

closed 2018-02-22 by Barbara1984

theyosh, 2018-02-22:
Cool! Updated the code. This value seems to work for me also

Barbara1984, 2018-02-22:
I am glad :-)