theyosh/TerrariumPI #84: SHT20 feature request

Aapjes94, 2018-01-29:
I'd like to put in a feature request for SHT20 weatherproof temperature and humidity sensor. I run a very humid environment and would like to monitor soil moisture as well. Other sensors easily get corroded under these circumstances, but these seem to be up to the task. The sensor is essentially just an SHT-20 with the 4 data/power wires brought out so any SHT-2X code for a microcontroller will work. It uses I2C, I'm not sure if that will form a problem.

theyosh, 2018-01-29:
Do you have an example of a device? I do find SHT15 and SHT21 but no 20. I do see some people got a SHT21 working on a PI. So that should be not an issue. But I would like to test it myself so I need some hardware to test with.

Aapjes94, 2018-01-29:
[I was thinking of something like this](https://www.makerfabs.com/index.php?route=product/product&product_id=384) or [this](https://www.aliexpress.com/item/2pcs-lot-Free-shipping-Temperature-and-humidity-sensor-module-shell-sht20-protective-cover-SHT10-soil-greenhouse/32756810803.html?spm=2114.search0104.3.9.5fc647b6y9WwEv&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10344_10068_10342_10343_10340_10341_10084_10083_10618_10304_10615_10307_10301_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_28,ppcSwitch_4&algo_expid=832f2e93-d9a3-4665-a0f4-3e119e5c0399-1&algo_pvid=832f2e93-d9a3-4665-a0f4-3e119e5c0399&priceBeautifyAB=3) The way I read it, it runs the same software as a SHT21, but is a cheaper(less accurate version).

theyosh, 2018-01-29:
Hmmm... they look cool... I will look into it.

theyosh, 2018-02-09:
Hardware is ordered for testing... shipping is about 20 days :(

Aapjes94, 2018-02-09:
Really cool that you're willing to do this, thanks a lot.

theyosh, 2018-02-26:
Just got update that shipment has started.... so still 20 days waiting :(

Aapjes94, 2018-02-28:
Thanks for the update. Chinese New Year delayed some of my orders as well.

theyosh, 2018-03-15:
Hardware has arrived!! Now we need to read Chinese :P

Aapjes94, 2018-03-15:
Cool! Let me know if you get stuck with some of the documentation and I'll try to help.

theyosh, 2018-03-16:
I figured out the colors with google translate to chinese.. So now it is only matter of connecting them and trying some code...

But if you have already something working, please share...

Aapjes94, 2018-03-16:
These are the colours as labelled from my order. Not using black for the ground is strange to me...

connection mode: VCC-red   GND-green  DATA-black  SCK-white

With studying and exams last week I haven't gotten around to testing the sensors myself but I found someone who linked [this](https://github.com/iotify/custom-lab/blob/master/sht2x.py) a while ago and saved that. I'm still learning python so while I feel comfortable editing code I'm not really able to write that well. This code is well documented and commented so it should provide a good starting point.

theyosh, 2018-03-16:
Ah, now it gets funny. I have the following colors: black, brown, yellow, blue :p

And I found: https://github.com/imonkey-tw/weather168_raspberry_pi/blob/master/SHT20/SHT2x2xively.py

So that are already two ways of coding. That is nice... I hope to do some testing this weekend

Aapjes94, 2018-03-16:
Let’s make this some more fun, shall we? I just looked it up and the product I have does not correspond with the product description. The sensors I have are RELed, Green, Blue and Yellow. 

I took the sensor appart and the PCB is labeled at least. 

Red - VCC
Green - GND
BLUE - SDA
Yellow - SCL

theyosh, 2018-04-01:
do a git clone, restart and clear browser cache. You should be able to add new type of sensors.

To find the right ID, use i2cdetect -y 1 and enter the found ID. Probably 40. Use that as address
Downside, only 1 sensor can be used for now. I do not know how change the sensor I2C address. That is needed to support multiple SHT2X sensors through I2C

BashSer, 2018-04-01:
How about "template" for i2c ? =D
User input bus (example, i use 1 for hw_clock, and 0 for other. Another can ise any multiplexer and i2c bus can = 3, 8 and etc), address and need formulas from datasheet. This can realization of template from GUI or import file where user write all needed data.

theyosh, 2018-04-01:
Sorry, it is unclear what you try to tell. I do not have I2C experience. Nor do I have other hardware to test. So please be more clear what you mean.

It is possible to enter a value like [BUS_ID],[HEX_ADDRESS] which can be 0,40 using bus 0 in stead of bus 1

BashSer, 2018-04-01:
[image]
*First use breadboard, sorry xD*
If you type "i2cdetect -y 1" you see only RTC module, but if you type "i2cdetect - y 0" you can see LCD, SHT15 and DHT12(i dont know about third module on this schema :D). But, again, user can use, example, MCP23017 and can use not only bus 0 and 1, but also bus 43 and etc.

In datasheet for DHT12 we can read:
```
"| BYTE ADDR | R/W |        Description       |           Note           |"
"|    0x00   |  R  | Humidity integral digits | Relative humidity values |" 
"|    0x01   |  R  |      Humidity Scale      | Relative humidity values |" 
"|    0x02   |  R  |  Temperature integer bit |   The temperature value  |" 
"|    0x03   |  R  |      Temperature scale   |   The temperature value  |" 
"|    0x04   |  R  |                          |         Checksum         |"
```
Ok, example code from internet for this:

```
  #read 5 bytes of data from the device address (0x05C) starting from an offset of zero
  data = bus.read_i2c_block_data(addr,0x00, 5)

  print "Humidity = " + str(data[0]) + "." + str(data[1]) + "%"
  print "Temperature : " + str(data[2]) + "." + str(data[3]) + "C"

  if (data[0] + data[1] + data[2] + data[3] ==  data[4]):
    print "checksum is correct"
  else:
    print "checksum is incorrect, data error"
```
And after this can be done "template" - user need input: I2C_BUS, ADDRESS and inner data.
Example:
```
BUS = 0;
Address sensor = 0x5C;
Hum = 0x00;
Hum_scale = 0x01;
Temp = 0x02;
Temp_scale = 0x03;
Check_sum = 0x04;
```
*80% sensor uses this data. Exception - more expensive sensors(temp + hum + pressure and other). But if user can import template and enter data himself, then this should not be a problem*

Many sensors have a programmed address on a bus, because it is not possible to use several identical sensors =(
But I heard user can use multiplexer and other things =)

For si7021 in short:
```
# 0xF5(245)	Select Relative Humidity NO HOLD master mode
bus.write_byte(0x40, 0xF5)
# Read data back, 2 bytes, Humidity MSB first
data0 = bus.read_byte(0x40)
data1 = bus.read_byte(0x40)
# Convert the data
humidity = ((data0 * 256 + data1) * 125 / 65536.0) - 6
 
#0xF3(243)	Select temperature NO HOLD master mode
bus.write_byte(0x40, 0xF3)
# Read data back, 2 bytes, Temperature MSB first
data0 = bus.read_byte(0x40)
data1 = bus.read_byte(0x40)
# Convert the data
cTemp = ((data0 * 256 + data1) * 175.72 / 65536.0) - 46.85
fTemp = cTemp * 1.8 + 32

# Output data to screen
print "Влажность: %.2f %%" %humidity
print "Температура C: %.2f" %cTemp
print "Температура F: %.2f" %fTemp
```
Or i can help to collect scripts for difference sensors =D

theyosh, 2018-04-02:
I am sorry, but above message is not helping with this issue. Please post on the right issues.
Also this will need very smart people, as even I am not able to understand this. How can this then be added by non technical persons. Sorry, this is not going to work.

This issue will be closed now, as support for the SHT20 is now available

BashSer, 2018-04-02:
If i give you tested and working python scripts for different sensors for measurment can you adding this in your project ?

theyosh, 2018-04-03:
Yes, but please, create new issue per sensor / code. So it is easier to debug and fix issues per sensor.

closed 2018-04-03 by theyosh