theyosh/TerrariumPI #86: Dashboard request

jbrooks4, 2018-01-29:
I was wondering about the dashboard average charts and was wondering if we could get a way to split them up based on sensors? ( group together just my 1wire sensors for a temp and another temp of the humidity ones) Or would this be easy to change myself? 

Cheers 
Justin

theyosh, 2018-01-30:
Hi Justin,

I am afraid that this goes beyond my idea of this software. Now the dashboard fits in 1 screen and is rather compact. If that has to be extended with more graphs it will grow a lot. Also I think that a dashboard should be clear, and not cluttered.

Also, this needs an interface to create custom graphs and selecting sensors. This will introduce new code, which has to be tested etc. And the fact that there can then 0 to unlimited graphs. This is now limited to one temperature and one humidity graph. So all the needed information is known upfront.
The next question will then be in which order should the graphs being shown. etc...

So I have to say that this will not be made by me. You are free to fork it and try it your self, but I do not see enough benefits for me to make it.

jbrooks4, 2018-01-31:
Okay, It was more of only 1 more chart to see my water temp, I have two 1wire waterproof sensors to know the water temp. I will see what I can do.

theyosh, 2018-01-31:
Ah, but that is somehow possible, I think.

If you put those sensors in the cooler part of the environment, and do not select switches, you will get an average temperature value at the dashboard at the cooler part of the environment...

You do not have graphs, but it will show the actual temperature, and update it every minute

closed 2018-01-31 by jbrooks4

kaju666, 2018-11-19:
Dear yosh is there any simple way to customize the dashbord? like for example adding webcam to it ?

theyosh, 2019-02-12:
@kaju666 Nope, the dashboard is static as is... I am not a UX designer, so I would like to keep it simple