theyosh/TerrariumPI #87: possible to add PH_meter(SKU__SEN0161) and HC-SR04-Ultrasonic-Sensor ?

nke69, 2018-02-05:
Good morning,

First and foremost thanks for this fantastic job that you made us enjoy;)

I modify it a little bit according to my needs as a freshwater aquarist :) I realize the French translation but it takes a little time.

I would like to add a PH probe like this but is it compatible with TerrariumPi?
here is his info:  https://github.com/jimaobian/DFRobotWiki/wiki/PH_meter(SKU__SEN0161)

And secondly, I would like to add an ultrasonic level sensor (http://www.instructables.com/id/HC-SR04-Ultrasonic-Sensor-With-Raspberry-Pi-2/)  but will he also be recognized by the program?

Thanks in advance,
nke69.

theyosh, 2018-02-05:
Thanks for the compliment.

Regarding the hardware. The Ph probe is an analog device which can't be readout by a Pi directly. So that needs an analog board in between. I am working on this, but it has a very low priority now.

The level sensor is something that could be easy added. But then the question rises what do you want to do with it? It does measure a distance. But does it need some kind of alarms?
Personal I am thinking to use it as a way to measure my spray water level. And when the distance is to big, the tank is empty and will just give an alarm.

But you could also think of something that trigger a water pump to fill tanks. That needs more data and actions. Also that needs some insurance it will not overflow etc...

So I will look into the level sensor, but the Ph probably not or somewhere this year. The analog part makes it also a bit harder to support. It needs more technical and electronic skills on a Pi.

nke69, 2018-02-07:
775/5000
Thanks for the clear answer;)

The PH probe is not a priority for me at this time and if somehow I can help I would love it;)

The level sensor meanwhile, my goal is to effectively calculate the water level available in the tank, past an alert threshold that I determined, the ideal yes would be to fill automatically via a pump perilstatique from a tank Reserve. I would pair it with an overflow sensor placed directly into the sensor aquarium that has a waterproof float (on / off) for safety. and which would disable the pump when the float determines the required height.

I'm still working on the best couple hc-sr04 or only the level sensor in concert with a pump

Jelly-Flinger, 2018-02-07:
It depends on how accurately you want to stop the pump at a given water level. You might be better to have 2 independent ultrasonic level sensors.  The reason being you can see that both sensors are operating as expected. A float switch you don't know if its working. (this is a common control philosophy for modern oil & gas facilities).  

I am guessing that the reservoir is small enough that if the tank reserve is accidentally  pumped out in its entirety that nothing bad will happen?

I have not personally looked into the accuracy or details of the ultrasonic meters you propose and don't know the details of your setup. Just some food for thought!

Aapjes94, 2018-02-07:
I'd love to see something like the ultrasonic level sensor as well. I'm currently using the software for controlling a small indoor greenhouse. An ultrasonic sensor could be used to track the growth of plants.

theyosh, 2018-02-08:
I will add support for the ultrasonic level sensor. But I think I will implement on a more generic way.

First it will be just a new kind of sensor.
Each sensor has a min and max limit. On which a trigger can be created
Multiple sensors can be combined to use the average values. This will not fix the issue of wrong hardware or faulty readouts.
Add a new environment part for the water tank.

The problem I have a bit with measuring plant height is that is probably inaccurate due to leafs. Also to know the height of the plant, there are more values needed. Like the height of the sensor. Else you cannot calculate the height. Because I measure distance from sensor to first object.

But you can still track the growth speed. Everyday the distance will be shorter, and that difference can be logged and graphed.

Also mixing this sensor with a switch kind sensor is harder to make. And because the heater, and cooler work already on the min and max values. So that is easier to expand.

theyosh, 2018-02-11:
Ultrasonic sensor is ordered...

theyosh, 2018-02-20:
Update: Followed: https://www.modmypi.com/blog/hc-sr04-ultrasonic-range-sensor-on-the-raspberry-pi

Used 10K and 4.7K Ohm resistors in stead of 1K and 2K. Does work.

[image]
[image]
[image]

Test code is working.... So I hope to have an early test version in a few days

nke69, 2018-02-21:
Hello, great job and thanks for following yosh :)

nke69, 2018-02-22:
Hello ;)
unable to run the hc-sr04 from the interface, I have this error every time:
2018-02-22 20: 00: 10,000 - INFO - terrariumWrapper - Starting TerrariumPI server at location: http://192.168.1.61:8090 ...
Traceback (most recent call last):
   File "/home/pi/TerrariumPI/terrariumPI.py", line 17, in <module>
     terrariumEngine = terrariumEngine ()
   File "/home/pi/TerrariumPI/terrariumEngine.py", line 88, in __init__
     self .__ load_sensors ()
   File "/home/pi/TerrariumPI/terrariumEngine.py", line 139, in __load_sensors
     self .__ UNIT_TYPE)
   File "/home/pi/TerrariumPI/terrariumSensor.py", line 51, in __init__
     self.sensor_address = {'TRIG': sensor [0], 'ECHO': sensor [1]}
IndexError: list index out of range

I have a basic python script from modmypi and when I run it it gives me the distance. I'm looking for but I'm drying on this error :(

in the settings if I put a hadwaretype other than hc-sr04, the ./start works and starts the application ..
On the other hand if I force the hardwaretype = hc-sr07 restart in loop and finally crashing of the soft

[Sensora4bfa7ca2c856bbccdbc39eff0102ecb]
address = 18 (18-20-23-24) always the same error
alarm = True
alarm_max = 25.0
alarm_min = 15.0
hardwaretype = hc-sr04
id = a4bfa7ca2c856bbccdbc39eff0102ecb
limit_max = 25.0
limit_min = 15.0
name = fleet
type = distance

theyosh, 2018-02-22:
Ah sorry, need to mention.
Input is 2 GPIO numbers, separated by ,

Using this setup: https://www.modmypi.com/blog/hc-sr04-ultrasonic-range-sensor-on-the-raspberry-pi

Enter first the value of TRIG,ECHO in the interface. It needs two GPIO ports to work on

nke69, 2018-02-22:
Perfect, the magic comma :)  thx a lot yosh

theyosh, 2018-02-22:
It does need some update in the documentation about this... :P

nke69, 2018-02-24:
in the log i've got always this error ;

UnboundLocalError: local variable 'pulse_start' referenced before assignment
pulse_duration = pulse_end - pulse_start
File "/home/pi/TerrariumPI/terrariumSensor.py", line 163, in update
Traceback (most recent call last):

Any suggestion ? knowing that my achievement is consistent with yours;)

theyosh, 2018-02-24:
sounds like you are measuring a small distance?
Made a small code change. Try git pull and restart

nke69, 2018-02-24:
Everything is ok now ;)

theyosh, 2018-02-24:
Ok... I will close this issue, and continue on the other one for the PH sensor.

closed 2018-02-24 by theyosh

nke69, 2018-04-27:
Hi,

Just tested now .. everything appear normal .. same pH on Raspberry and on Arduino ;)

theyosh, 2018-04-27:
Cool. I updated the code for remote Ph as mentioned by you above. So then this is all fine again. Thanks for testing!

javadth, 2019-08-18:
[دوربين مداربسته داهوا](https://xn----zmcbckzcnev7qkavc6a.com)